google/openhtf #516: Create prompt plug that only displays (no text input)

wallacbe, 2017-02-17:
Create prompt plug that only displays (no text input)

fahhem, 2017-02-25:
This request boils down to splitting the current `UserInput` plug into two: one that displays text, and another that requests text input. A parallel request was to create a third plug that accepted key-presses directly, rather than line-buffered input.

Kenadia, 2017-02-27:
What's the difference between this requested feature and using the current UserInput plug with text_input=False?

fahhem, 2017-02-28:
I think the current UserInput implementation ignores text_input and always shows the textbox and requires a response, right? If that's not the case, or is fixed, then it would be equivalent (except for the separate key-press-input plug)

Kenadia, 2017-02-28:
This is what we have currently

https://s23.postimg.org/bakn898zv/Untitled.png
(having trouble with GitHub uploads)

Example script:

```python
import openhtf as htf
from openhtf.plugs import user_input

@htf.plugs.plug(user_input=user_input.UserInput)
def prompt(test, user_input):
  """A hello world test phase."""
  result = user_input.prompt('Hello! Press OK to continue.')

if __name__ == '__main__':
  test = htf.Test(prompt)
  test.execute(test_start=lambda: 'my-dut-id')
```

Kenadia, 2017-02-28:
@wallacbe can you comment on whether this meets your needs?

wallacbe, 2017-02-28:
I'm looking for ability to create a plug that overrides existing prompts and does not require OK button in UI.

1. Displays "Press P to pass or F to fail"
2. Poll keyboard for key press (or whatever input mechanism, could be a GPIO, etc..)
3. Clear prompt

glados-verma, 2022-10-13:
Closing as stale.

closed 2022-10-13 by glados-verma