google/openhtf #522: Add need pass property for measurements

wallacbe, 2017-02-24:
Add need pass property for measurements

Sometimes a measurement passing is a dependency for future tests.  Would like to make measurement configurable such that openhtf.PhaseResult.STOP would be returned and test would fail if measurement did not pass

fahhem, 2017-02-25:
How often does this occur? It's pretty rare for us, so we explicitly return htf.PhaseResult.STOP for those instances.

wallacbe, 2017-02-25:
I have several cases of this.  I'm setting limits as constants, checking max/min condition, then returning openhtf.PhaseResult.STOP, which can get verbose.

This would also allow setting up testers as "production" mode and "debug" mode via conf where we can fail tests immediately versus continuing.

wallacbe, 2017-07-27:
Duplicate issue https://github.com/google/openhtf/issues/619

closed 2017-07-27 by wallacbe