google/openhtf #574: Run Tests over the WebGui

jonas-steinkamp, 2017-05-12:
Is it possible to restart a test over the web_gui?

I want to setup a station with a test for a pcb. The OpenHTF script runs on a Raspberry PI. The test operator has another pc on whitch he opens a browser and starts the test.
If one PCB has passed can i restart the test from the web_gui or how is the concept for that?

And one other thing is it possible to extend the station api to get a progressbar for every  #phase?

Thank you and have a wonderfull day,
Jonas

fahhem, 2017-05-12:
The expected case here is to run your test in a loop and have it block on the web_gui for starting each run. You can loop using a `while True:` loop in Python or a `while true; do ... done` loop in bash, as long as the station_api_port doesn't change between runs your web_gui will show up.

If you use the built-in `test_start` or another one that uses `user_input`, then you can loop the test and it will wait until the operator interacts with the station to start. The recorded start time of the test is after `test_start` finishes

jonas-steinkamp, 2017-05-19:
Thank you @fahhem for your fast answer. I will try it out.

Have a nice day,
Jonas

closed 2022-10-13 by glados-verma