google/openhtf #578: @test.yields_phases and @conf.save_and_restore need to be switched in docs

arsharma1, 2017-05-13:
I believe the docstrings that say:
>     # Decorate the test* method with this to be able to yield a phase to run it.
>     @test.yields_phases
>     # Decorate with conf.save_and_restore to temporarily set conf values.
>     @conf.save_and_restore(phase_variance='test_phase_variance')
>     def test_first_phase(self):
>       pass  

should be reordered with the conf.save_and_restore decorator first.  Otherwise, the conf value is not changed when running the unit test case. 

Is this expected and the docs out of date?

fahhem, 2017-05-13:
Hm, why is it not changed while the test runs? Does `yields_phase` not preserve the conf's wrapped function?

arsharma1, 2017-05-13:
I think it's because the unit test function is a generator.  Calling the generator function returns the generator iterator immediately, so the conf restoration occurs before anything in the generator runs.

closed 2022-10-13 by glados-verma