google/openhtf #581: Progressbar in Phases

jonas-steinkamp, 2017-05-22:
I want to use the openhtf framework to automate the line testing of PCBs.

One of the processes is that a i have to update the firmware for the micro controllers.
This process takes some time and the user can't see any progress and is now wondering about goes every thing well or not.

A solution for this problem would be a progressbar for the phase.

I have added a progressbar plug to send data to the web frontend. And on the frontend i have added a new component, that displays the bar. But i think this is a little hack and the data is not on the right place.

``` javascript
@Component({
  selector: 'progressbar',
  templateUrl: 'templates/station.progressbar.html',
  styleUrls: ['styles/station.progressbar.css'],
  directives: [],
  providers: [StationService]
})

export class ProgressBar {
  private _progressbar: any;

  @Input() set progressbar(value: any) {
      for(var plug in value){
        if(value[plug] && value[plug].hasOwnProperty("progressbar")){
          this._progressbar = value[plug]["progressbar"];
        }else{
            this._progressbar = null;
        }
      }
  }
  get progressbar(): any {
      return this._progressbar;
  }
```
The  `for(var plug in value){ ... }` is a hack so that i can inherit the progress plug to other plugs.

If someone has a idea, it would be great to hear it.
Thank you and have a nice day.

jettisonjoe, 2017-05-22:
Progress bars for firmware flashing and similar would be awesome. Here are some thoughts:
  * Phase-level progress bars are likely too coarse, as many use cases will involve flashing multiple devices/images in one phase. We'll probably want progress bars for smaller, more specific scopes than a phase.
  * Web GUI widgets for plugs is already something we've wanted for a while. A progress bar sounds like exactly the kind of custom widget that could fit nicely into a plug widget.
  * Design-wise, I'm generally foggy on how to nicely display plug widgets in the web GUI. I wish we had a UI designer in the mix. Open to ideas.
  * As an intermediate solution, logging can be used to display progress with no change to OpenHTF itself. Just make your test script more verbose with the logging and it should show up in the web GUI.

fahhem, 2017-05-22:
I think there are two things here. Jonas is asking about his UI plug, and Joe is talking about adding it to OpenHTF directly.

Jonas: If you want to support subclasses of plugs, the options we've looked at include:
* Maintain a list of all supported subclasses, which is fine when you have a limited few.
* Maintain some way of determining if it's a supported plug, such as what you're doing.
* Change OpenHTF to pass along all base classes along with the class's name.

Joe: I think the right way for now is to keep it as a downstream/separate plug until we have some idea on the APIs being used and how often.

jonas-steinkamp, 2017-05-23:
**Thank you for your answers. :)**

@fahhem 
Like @jettisonjoe i think also it would be a good idea to add the function direct to the phase.

**For example:**
Extend the TestApi with a progress property.
- Default is None, no progressbar is shown.
- 0 - 100 the progressbar shows the percent value.
- If a phase completed it would be set to None again.

So that a phase can look like this:
```python
def example_phase(test):
    for value in range(100):
        test.progress = value
```

I have pushed my [progressbar](https://github.com/jonas-steinkamp/progressbar_plug) to github today if someone will have a look.

fahhem, 2017-05-24:
I'm not against pulling in a progress bar plug, but keep in mind there are many ways of doing this. If it goes into `TestApi`, a lot more thought has to go into its usage and what operators want out of it. They presumably want a test-wide progress bar, rather than single phases.

If we consider the file-copy progress bar example, then having a progress bar that moves to the right smoothly is the ultimate goal. I'd consider two progress bars, one for the test and one for the phase (or current operation rather) as being another acceptable answer. A progress bar that comes and goes though is a very frustrating experience, because when you look at the screen you have no indication how far you are from done. Operators want quick-glance method of determining whether they need to interact with the station or wait, and roughly how long to wait.

If you wanted to push your progress bar plug into `contrib/plugs/` I'd be happy to handle the code review. We haven't yet solved the problem of not needing to modify the frontend to handle new plugs, so enabling it in the UI would have to remain as a separate integration task for now.

jo-nas, 2017-05-24:
I can understand you opinion about the progress bar. For this time i will use my plug.

I can push a markdown file with links for my plugs in this folder. But considering testing i would prefer to have it in a own repository. Also i think there will be some changes in the next time. I am working on it.

closed 2022-10-13 by glados-verma