google/openhtf #598: Running frontend and tester raises TypeError: __new__() takes exactly 8 arguments (7 given) in frontend

wallacbe, 2017-06-22:
Starting openhtf web gui server on http://localhost:12000.
Exception in thread Thread-1:
Traceback (most recent call last):
  File "/usr/lib/python2.7/threading.py", line 810, in __bootstrap_inner
    self.run()
  File "openhtf/output/web_gui/__init__.py", line 247, in run
    self._handle_stations(self.stations.iterkeys())
  File "openhtf/output/web_gui/__init__.py", line 231, in _handle_stations
    self.watch_tests(hostport)
  File "openhtf/output/web_gui/__init__.py", line 259, in watch_tests
    hostport, remote_test, self._on_update_callback)
  File "openhtf/output/web_gui/__init__.py", line 139, in __init__
    self.handle_state_update()
  File "openhtf/util/threads.py", line 138, in synchronized_method
    return func(self, *args, **kwargs)
  File "openhtf/output/web_gui/__init__.py", line 153, in handle_state_update
    state = self._test.state
  File "openhtf/core/station_api.py", line 123, in _wrapper
    return func(*args, **kwargs)
  File "openhtf/core/station_api.py", line 436, in state
    remote_state_dict and RemoteState(cached_state, **remote_state_dict))
  File "openhtf/core/station_api.py", line 314, in __new__
    running_phase_state and RemotePhase(**running_phase_state))
TypeError: __new__() takes exactly 8 arguments (7 given)

wallacbe, 2017-06-22:
If I remove this commit it works:

commit 4dbaccd8140a7dce4b3640d4eb1646536d3c0066 (HEAD, origin/master, origin/HEAD, master)
Author: Ken Schiller <Kenadia@users.noreply.github.com>
Date:   Wed Jun 21 17:03:11 2017 -0700

    Always remove attachment data when converting to dict (#597)

fahhem, 2017-06-24:
Ah, looks like `TestState._asdict` lost the `measurements` field in that commit, likely on accident?

closed 2017-06-24 by fahhem