google/openhtf #604: Don't initialize plug in skipped phase

wallacbe, 2017-06-28:
Observed that when run_if=false for a phase the plug was still initialized (only phase using the plug).  If a phase is disabled with dependent equipment we shouldn't require the equipment to be present.

jettisonjoe, 2017-06-28:
Tricky, since run_if functions are run at phase execution time, whereas plug initialization happens early in test execution, before phase execution. Maybe we can collectively figure out a workaround for your particular case if you can provide a bit more of an example. What is your run_if condition, and can it be checked before execution to simply avoid adding the phase to the test instead?

wallacbe, 2017-06-28:
Let's say I have a phase that measures temperature with a piece of equipment that enumerates as a serial port over USB.  The plug constructor opens the serial port, so if not present it will error out.  We some times pass a config value in the constructor indicating the iSerial, but that's not always the case (sometimes auto detect with VID and PID).

run_if is a boolean config to enable/disable temperature phase

fahhem, 2017-06-28:
Doesn't run_if take no arguments? Then we could call them all before the
test runs and replace phases that return false with a SKIP phase with no
plugs

jettisonjoe, 2017-06-29:
Yes we could, but it does change the existing behavior of run_if. Before the change, it will have been possible to create run_if functions that depend on state only present during test execution (for better or for worse).

Replacing skipped phases with a special SKIP phase would also change the expected output from what it is now, as I believe skipped phases currently still enter the test record. Replacing them with a special skip phase would alter the way they appear in the finished record when skipped.

Not saying we _shouldn't_ do the change, but it's worth pointing out that it's not a trivial internal-only change, and will have user-facing implications we should consider.

glados-verma, 2022-10-13:
Users are advised to use run_if only for decisions made during a test, and that during a test they should expect all plugs to be instantiable. Otherwise remove phases from the list of phases before starting the test.

closed 2022-10-13 by glados-verma