google/openhtf #613: empty codeinfo in json output

DanLipsitt, 2017-07-25:
The output of `examples/hello_world.py` has multiple empty codeinfo sections (and none that are populated). They look like this:

```json
  "code_info": {
    "docstring": null,
    "name": "",
    "sourcecode": ""
  },
```

I'm using OpenHTF v1.1.0.

fahhem, 2017-07-25:
Yes, by default we do not capture source anymore. Your config file must set `capture_source: true` (or you must set conf.capture_source to `True` some other way before calling `openhtf.Test()`) and then it captures the source.

This decision was made to avoid spending the cost, at startup, of capturing source code (which can take significant time in some cases) and to err on the side of not capturing since, in larger firms, the source code leaking to suppliers is a (bureaucratic) worry.

closed 2017-07-25 by fahhem

DanLipsitt, 2017-07-25:
Excellent, thanks. Is there some documentation about this that I missed? If not, can I start some?

This came up, by the way, because I was trying to figure out if there is anything that uniquely identifies a test phase in the output. It seems like it's easy to end up with test results in the database that are hard to differentiate and query.

fahhem, 2017-07-26:
We'd be happy if you start a docs/ folder :)

If the phase's name isn't unique, there should be a phase uuid called `descriptor_id` that's unique per run of the binary.

DanLipsitt, 2017-07-26:
What I'm looking for is a unique test/phase key for database queries. One that is the same across test runs. Apologies if this is a bad place to ask about this.