google/openhtf #615: phase_result is CONTINUE for phases that fail validation

DanLipsitt, 2017-07-25:
Why is does `phase_result` equal `CONTINUE` rather than `STOP` for phases that have a measurement that fails validation? The next phase does not run.

Example output excerpt from a simple [script](https://github.com/DanLipsitt/openhtf-experiments/blob/master/fail.py):
```json
{
  "outcome": "FAIL",
  "end_time_millis": 1501004966765,
  "options": {},
  "attachments": {},
  "name": "phase2",
  "descriptor_id": 4337164368,
  "measurements": {
    "x": {
      "outcome": "FAIL",
      "measured_value": -1,
      "name": "x",
      "validators": [
        "0 <= x <= 1"
      ]
    }
  },
  "start_time_millis": 1501004966764,
  "result": {
    "phase_result": "CONTINUE"
  },
  "codeinfo": {
    "docstring": null,
    "name": "",
    "sourcecode": ""
  }
}
```

mattvenn, 2019-01-11:
I'm using a validator and I don't want have to repeat the validators again inside the function to test if it should be repeated

```
@htf.PhaseOptions(repeat_limit=10) # timing dependent on when the power comes up, so check lots of times                                                                                                                                                  
@htf.measures(htf.Measurement('cksum').equals(FW_CKSUM)) # should be the same                                                                                                                                                                             
@htf.measures(htf.Measurement('version').equals(1))                                                                                                                                                                                                       
@htf.plug(side_plug=LeftPlug, power_plug=PowerPlug)                                                                                                                                                                                                       
def read_left_bootloader_version(test, side_plug, power_plug):                                                                                                                                                                                            
    power_plug.power_off()                                                                                                                                                                                                                                
    power_plug.power_on()                                                                                                                                                                                                                                 
    version, cksum = side_plug.read_bootloader_version()                                                                                                                                                                                                  
    test.measurements['cksum'] = cksum                                                                                                                                                                                                                    
    test.measurements['version'] = version                                                                                                                                                                                                                
    test.logger.info("cksum: %s version: %s" % (cksum, version))                                                                                                                                                                                          
    if version != 1:                                                                                                                                                                                                                                      
        return openhtf.PhaseResult.REPEAT                                                                                                                                                                                                                 
    if cksum != FW_CKSUM:        
        return openhtf.PhaseResult.REPEAT                                                                                                                                                                                                                 
```

arsharma1, 2019-01-11:
To Dan's question, measurements cause individual phases to pass or fail, but cannot cause the test run to stop.  The common way I handle that is by adding a gate phase that stops the test if any previous phases failed.

For Matt, we don't have a good solution to this yet.  I thought about having an htf.PhaseResult for REPEAT_IF_MEASUREMENTS_FAIL, but this would probably start increasing those unsustainably.  The other option would be to add such options when defining the htf.Measurement.  I'll work on a feature request for this.

arsharma1, 2019-01-18:
We determined that this information is already available, but wasn't documented very well.  One can get access to the measurement status with the TestApi.get_measurement function:
```
@htf.measures(htf.Measurement('a').with_validator(my_complex_validator(INPUTS))
def phase(test):
  # ... stuff
  if test.get_mesurement('a').outcome != measurements.Outcome.PASS:
    return htf.PhaseResult.REPEAT
  else:
    return htf.PhaseResult.CONTINUE
```

This keeps the result logic entirely embedded inside the phase function without making the measurements harder to define.  It also gives the author the best control on when to make these sorts of decisions.

closed 2019-01-18 by arsharma1