google/openhtf #617: User input specification

DanLipsitt, 2017-07-26:
Is there a recommended DRY way to parse and validate user input? Could the UserInput plug somehow share validation with a measurement?

Maybe #415 could help with this?

jettisonjoe, 2017-07-26:
I guess it depends on what the end goal is, but I would probably just do it the same as if I were validating raw input in native Python. i.e. import whatever library you want and use it in the body of the phase. I would probably just put a loop around that prompt.

As a more elegant solution, I would definitely entertain the idea of attaching validators to the prompt call, then you can use the same validator in decorating the phase and in sending the prompt. I agree that we probably want to  move forward on #415 before going too nuts with validators on prompts.

Even with validators in the prompt, getting that to do something _nice_ in the web frontend rather than just re-displaying as if it's a new prompt would require more work depending on how we would decide to do it.

fahhem, 2017-07-27:
Internally, we have quite a few helper functions written around user inputs. I've tried coalescing them, but there are so many different interactions that it's difficult. In some cases, you want to add a specific error message to the prompt and show it immediately, in others you want to retry the whole phase, etc. A function could likely be written to handle all of these, but it would take 5+ parameters and be much more confusing to everyone over purpose-written ~5 line functions.

I agree with Joe, validators in the prompt is the easy half of the battle. What to do in response to that validation is very much in the air. We do accept code into contrib/ if you have some helper functions you'd like to share. Ours are full of strings specific to each test or I'd share them myself.

DanLipsitt, 2017-08-01:
Do you have a simple example of what adding an error message to the prompt looks like?

fahhem, 2017-08-09:
Here's an example that uses a validator:
```py
checker = validators.in_range(0, 5)
base_msg = msg = "Enter a number between 0 and 5"
error = "Please retry"
while True:
  result = user.prompt(msg)
  if result.isdigit() and checker(int(result)):
    break
  msg = base_msg + "\n" + error
```

Internally we just wrote a UserInput subclass with an `error` attribute for the UI. We may upstream that if it does what we need.

closed 2022-10-13 by glados-verma