google/openhtf #620: Working without validators

DanLipsitt, 2017-07-27:
Because validators seem not to have an option to abort the rest of a test when they fail (#619), my team is considering abandoning validators and doing manual validation within phase bodies. What would we miss out on by doing this?

fahhem, 2017-07-27:
We have a station UI internally that we want to push upstream to openHTF that uses the validators to similarly describe the data. I believe the one in openHTF also uses the validators.

We also have an internal cloud-based UI (for uploaded and vendor results) that uses the validators to describe the data, though that won't be available openly for a very long while (though enough clamoring might convince higher-ups? :)

Essentially, what you're missing out on is the UI, though you're missing out on it at the moment anyway. Otherwise, they're only used for marking a measurement, and therefore a phase as a failure.

closed 2017-07-27 by fahhem

DanLipsitt, 2017-07-27:
Clamor clamor! 😉

sciguy14, 2017-08-09:
@fahhem is there anybody in particular whom I can clamor to?

fahhem, 2017-08-13:
If you're willing to pay for it, then I'll ask what it would cost to expose and go from there.