google/openhtf #622: compare measurements in a validator

DanLipsitt, 2017-07-27:
Is it possible to validate measurements against each other, either within or between phases?

For example, validate that measurement1 and measurement2 are equal, after determining measurement1 at runtime.

It seems that validators have no access to context that would allow them to do this.

fahhem, 2017-07-28:
No, in these cases we usually add a third measurement that validates
against True and do the equality check in the phase. I've only seen this in
a few rare cases, usually there's a constant number or range being
validated against that's available at import time (python's 'compile time')

DanLipsitt, 2017-07-28:
I don't think I understand the use of the third measurement.

Our use case is burning a serial number to a new DUT:
- Get the serial from the DUT.
- Validate that it's blank.
- Burn a serial pulled from a barcode.
- Validate that the serial on the DUT matches the barcode.

kdsudac, 2017-07-28:
It's not very elegant but the way we currently handle would be something like this:

```
test.measurement.serial_matches = barcode_serial == burned_serial
```

It's an extra measurement, but measurements are free (aside from a little extra clutter).

kdsudac, 2017-07-28:
For the record, I can see some use cases for having measurement validation between phases (your serial number example is a good one).

I think the tricky part is when you start mapping that into a manufacturing setting it can lead to bad habits like dynamic specifications.

fahhem, 2017-07-28:
On top of kdsudac's record, we do allow for some amount of dynamic
specifications. For example, we have placeholder validators that are
replaced at runtime based on the board revision, but that's still before
openHTF sees the validators. Once openHTF starts running the phases, the
measurements are static. Once we hammer out the corner cases of the
revision-based validators, we'll be upstreaming them as well, in case
you're interested.

DanLipsitt, 2017-07-28:
Thanks for the example, @kdsudac.

As for bad habits: this is a valid use case that we need. The question is whether we can do it in a structured way using validators or not.

DanLipsitt, 2017-07-28:
Hmm, my last comment came off sounding snippy, which I didn't intend. Just trying to navigate some competing design goals.

kdsudac, 2017-07-28:
I didn't interpert as snippy :).

Can you share your use case? (or is it the serial burn-in case you mentioned above).  Good to get feedback on how users are actually using openhtf.

DanLipsitt, 2017-07-28:
Yeah, the serial burn in.

DanLipsitt, 2017-07-28:
Mockup [code](https://github.com/DanLipsitt/openhtf-experiments/blob/master/burn_serial.py).

kdsudac, 2017-07-28:
Created https://github.com/google/openhtf/issues/623 to bring in some logic that makes fetching measurements and attachments a little bit easier.

It won't do exactly what you want, but it'll make this line a little more sane:
```
scanned_serial = test.test_record.phases[-1].measurements['scanned_serial'].measured_value
```
will become something like:
```
scanned_serial = test.get_measurement('scanned_serial')
```

fahhem, 2017-07-29:
So does this pattern work for you?

```py
@htf.measures(htf.Measurement('burnt_serial_matches').equals(True))
def CheckSerial(test):
  expected = test.get_measurement('scanned_serial').value
  burnt = ....
  test.measurements.burnt_serial_matches = expected == burnt
```

DanLipsitt, 2017-08-01:
@fahhem Yes, that works. Keeping an eye on #623 and #624 with interest as well.

andrewwade, 2018-04-04:
Any update on a clean way to do this? I'm using this approach.
`@htf.measures(htf.Measurement('serial_number').doc('Read Serial Number'))
def test_read_serial_number(test):
    test.measurements._measurements['serial_number'].with_validator(validators.Equals(test.dut_id))
    test.measurements.serial_number = get_serial_number()`

glados-verma, 2022-10-13:
OpenHTF now has Diagnoses which can be used for this purpose.

closed 2022-10-13 by glados-verma