google/openhtf #623: Upstream measurement/attachment retrieval into TestApi

kdsudac, 2017-07-28:


DanLipsitt, 2017-08-01:
I don't want to add too much noise to PR #624, so I'll ask here: 

@fahhem [said](https://github.com/google/openhtf/pull/624#issuecomment-319463318)
>Phase names don't have to be unique either, and phase uuid's change until the test starts, and you can reuse the same phase twice in a row so there's no possible way to uniquely identify a phase. Instead of dealing with all of that, we pushed the burden back and said if you need to retrieve a specific measurement, name it uniquely. If you reuse a phase, use .with_args to give each an index or name.

How does `with_args` help here?

fahhem, 2017-08-02:
If you have a phase repeating, here's what you can do:

@htf.measures(
    htf.Measurement('voltage_attempt{attempt}'),
)
def Phase(test, attempt):
  test.measurements['voltage_attempt%s' % attempt] = 9

htf.Test(Phase.with_args(attempt=1), Phase.with_args(attempt=2))

Or:

VOLTAGE_MEAS = htf.Measurement('voltage_attempt{attempt}')

@htf.measures(VOLTAGE_MEAS)
def Phase(test, attempt):
  test.measurements[VOLTAGE_MEAS.name] = 9

htf.Test(Phase.with_args(attempt=1), Phase.with_args(attempt=2))

fahhem, 2017-08-02:
If you have a phase repeating, here's what you can do:

```py
@htf.measures(
    htf.Measurement('voltage_attempt{attempt}'),
)
def Phase(test, attempt):
  test.measurements['voltage_attempt%s' % attempt] = 9

htf.Test(Phase.with_args(attempt=1), Phase.with_args(attempt=2))

Or:

VOLTAGE_MEAS = htf.Measurement('voltage_attempt{attempt}')

@htf.measures(VOLTAGE_MEAS)
def Phase(test, attempt):
  test.measurements[VOLTAGE_MEAS.name] = 9

htf.Test(Phase.with_args(attempt=1), Phase.with_args(attempt=2))
```

DanLipsitt, 2017-08-02:
Even though phases can repeat, they do help narrow down the scope when you're looking for a measurement a lot. You could have a ton of phases with a measurement called `voltage` and as long as the phases had different names you'd be fine. Here's what I just cobbled together:

```python
def get_measurement(test, phase_name, measurement_name):
    """Look up a measurement from a previous phase by name.

    Args:
        phase_name(str): The requested phase.
        measurement_name(str): The requested measurement.

    Throws:
        LookupError: When the requested measurement or phase is not found or
        has not been set.

    Returns:
        The value of the requested measurement from the most-recent run of the
        requested phase from the current test. Phases can be run more than
        once in given test.

    """
    try:
        phase = [p for p in test.test_record.phases
                 if p.name == phase_name][-1]
    except IndexError:
        raise LookupError('Phase "{}" not found.'.format(phase_name))
    try:
        measurement = phase.measurements[measurement_name]
    except KeyError:
        raise LookupError('Measurement "{}" not found in phase "{}"'.format(
            measurement_name, phase_name))
    try:
        value = measurement.recorded_value
    except AttributeError:
        raise LookupError('No value for measurement "{}" in phase "{}"'.format(
            measurement_name, phase_name))
    return value
```

closed 2017-12-27 by kdsudac