google/openhtf #628: Have unrun phases show up in test record

wallacbe, 2017-07-28:
Feature Request
Have unrun phases show up in test record.  Have measurements and outcomes be UNSET.  Gives output callbacks and frontend added flexibility

fahhem, 2017-07-29:
Would the phase's outcome be SKIP here, or a new one like NOT_RUN? The frontend should have access to the test descriptor, which has all the phases. For output callbacks, you could get your test descriptor from `test.descriptor` and use that. There are UUID's attached to each phase to be able to detect which ones ran and which didn't. Is that a good enough answer?

wallacbe, 2017-07-29:
NOT_RUN sounds like a good idea.  For my output callback I'm inheriting from OutputToFile and I only see test_record, which doesn't contain all phases if tester exited early due to an ERROR

fahhem, 2017-07-29:
You could override `__init__` and take in the descriptor there.

wallacbe, 2017-07-30:
That approach worked and phases/measurements in the descriptor were in the same order as the test record.

fahhem, 2017-07-30:
Sounds like it's acceptable? Closing the issue then

closed 2017-07-30 by fahhem

wallacbe, 2017-07-30:
Ya, acceptable, thanks