google/openhtf #630: GUI doesn't update if test run ends quickly

DanLipsitt, 2017-08-01:
Here, for example, we can see a six-second delay before the GUI logs a request from the test script:
```
python -m openhtf.output.web_gui > /dev/null &
$ date; python openhtf/examples/hello_world.py
Tue Aug  1 15:53:29 PDT 2017
Enter a DUT ID in order to start the test.
192.168.10.184 - - [01/Aug/2017 15:53:35] "POST /RPC2 HTTP/1.1" 200 -
```

If we enter a DUT immediately on the console (or run a script that don't require user input and exits quickly) then no test results ever show up in the GUI.

DanLipsitt, 2017-08-01:
For now, I'm wondering if I can trigger `notify_update` from a `teardown_function`.

jettisonjoe, 2018-05-05:
Is this still an issue with the new web GUI?