google/openhtf #633: Code style problems

DanLipsitt, 2017-08-07:
The [contributor guidelines](https://github.com/google/openhtf/blob/c7da2958a8/CONTRIBUTING.md) for this repo ask contributors to ensure lint "still" passes before submitting a PR. However, pylint output contains many errors and warnings, as well as some crashes (see #632).

Additionally, the codebase uses two spaces for indentation throughout even though the contribution guide says to follow the Google Python Style Guide, which [specifies four spaces](https://google.github.io/styleguide/pyguide.html?showone=Indentation#Indentation) (as does [PEP8](https://www.python.org/dev/peps/pep-0008/#indentation)). This was discussed earlier in #188 but not resolved.

fahhem, 2017-08-08:
Yes, this code was exported from internal, where the style guide used to specify 2 spaces (but now allows either, though expects consistency on the project). The unfortunate case of a wholesale whitespace change is that it causes git blame and all in-progress work to break in functionality, and provides very little benefit when everything is otherwise consistent.

Regarding the lint, it seems like the lint plugins were written, but since Travis doesn't check them they quickly bitrotted. You can ignore the plugins until they're fixed/hooked up to Travis

closed 2017-08-08 by fahhem

DanLipsitt, 2017-08-08:
`git blame` has an ignore-whitespace flag (`-w`). Unfortunately it doesn't look like github does.