google/openhtf #635: outcome_details not set for aborted test run from test_start

DanLipsitt, 2017-08-07:
Here's some simplified code that aborts inside a phase run from `test_start`:

```python
import openhtf as htf
from openhtf.output.callbacks import console_summary

@htf.PhaseOptions(repeat_limit=1)
def repeat(test):
    test.test_record.dut_id = '1'
    return htf.PhaseResult.REPEAT

if __name__ == '__main__':
    test = htf.Test()
    test.add_output_callbacks(console_summary.ConsoleSummary())
    test.execute(test_start=repeat)
```

In real life, this would show up if we are trying to get a DUT ID from user input and repeat if it's invalid.

The code triggers a bug when it gets to `ConsoleSummary`:

```
2017-08-07 15:51:04,454 - ERROR - Phase returned REPEAT, exceeding repeat_limit.
Finishing test with outcome ABORTED.
Finishing test execution early due to test abortion, outcome ABORTED.
2017-08-07 15:51:04,457 - ERROR - Output callback <openhtf.output.callbacks.console_summary.ConsoleSummary instance at 0x1026d0a28> raised; continuing anyway
Traceback (most recent call last):
  File "/Users/dan/src/shaper/openhtf/openhtf/__init__.py", line 271, in execute
    output_cb(final_state.test_record)
  File "/Users/dan/src/shaper/openhtf/openhtf/output/callbacks/console_summary.py", line 70, in __call__
    (self.indent, record.outcome_details[0].code))
IndexError: list index out of range
```

I have a patch for `ConsoleSummary` that prevents this error, but before I submit it I want to ask whether it makes more sense to fix the underlying issue, which is that `record.outcome_details` isn't being populated.

fahhem, 2017-08-08:
Hmm, where is this path that gets around `outcome_details`?

fahhem, 2017-08-08:
I mean, yes, that's probably the right fix.

DanLipsitt, 2017-08-08:
I think it's because running a phase from `test_start` isn't fully fleshed-out. I haven't done a dive into that part of the code, though.

fahhem, 2017-08-08:
Yeah :/ `test_start` isn't quite a first-class citizen yet, I wouldn't be surprised if it's funky with this sort of stuff

DanLipsitt, 2017-08-08:
Can you give me some insight into the intended behavior of `test_start`? It would help me figure out what to do about this and #636. I don't know if I understand the inner workings of openHTF well enough to fix this.

fahhem, 2017-08-09:
Here's roughly how I view the test start procedure:

* Initialize minimal plugs for first phase
* Run `test_start` phase
* Set test start timestamp
* Initialize all plugs
* Run remaining phases
* Run teardown phase
* Set test stop timestamp

Any step that fails/stops should end the test and set outcome details. If a 'normal' phase (after `test_start`) has failed/stopped, then short-circuit to the teardown phase. We *should* always set the test stop timestamp, no matter why the test failed/stopped/errored.

PR #637 (2017-08-09): Populate outcome_details when test_start fails.
Make sure `outcome_details` gets populated from a `test_start` phase, even if it aborts.

I don't know how to unit test this, but here's an end-to-end test:
```python
import openhtf as htf

@htf.PhaseOptions()
def phase(test):
    test.test_record.dut_id = '1'
    return htf.PhaseResult.STOP

def check_outcome(record):
    assert(record.outcome_details)

if __name__ == '__main__':
    test = htf.Test()
    test.add_output_callbacks(check_outcome)
    test.execute(test_start=phase)
```

Depends on #626. Fixes #635.