google/openhtf #636: Better handling of BlankDutIdError

DanLipsitt, 2017-08-07:
Currently, if the phase run from `test_start` fails or aborts, the process exits with a traceback and this error:
```
openhtf.core.test_state.BlankDutIdError: Blank or missing DUT ID, HTF requires a non-blank ID.
```
No output callbacks are run.

We want to have normal output from this phase. For example, if a barcode scanner is malfunctioning, or a plug is failing to connect to a DUT, we want a record of the fault.

fahhem, 2017-08-08:
There are a ton of places where we assume there's a DUT serial, including many output callbacks. I'm not against this, but we'd need to clean up anywhere that's operating on the serial that could get possibly get a blank serial (but not everywhere).

DanLipsitt, 2017-08-08:
I don't think the system needs to be able to handle a blank DUT serial in general, we just need to get results out rather than crashing if it fails due to a blank DUT.

DanLipsitt, 2017-08-08:
Description updated to reflect the fact that it doesn't matter whether the failure occurs in `test_start` or not.

fahhem, 2017-08-09:
So if `test_start` doesn't set the dut serial, and one of your phases exits, there's no output callbacks called?

DanLipsitt, 2017-08-10:
Right. The most basic example is this:

```python
import openhtf as htf
from openhtf.output.callbacks import console_summary, json_factory

test = htf.Test()
test.add_output_callbacks(
    json_factory.OutputToJSON('output.json', indent=4),
    console_summary.ConsoleSummary())
test.execute()
```

This is a silly do-nothing setup, obviously, but it shows the behavior. No console summary, no output.json. Just the `BlankDutIdError` traceback.

jettisonjoe, 2018-05-04:
This issue is now completely avoided by way of configurable default DUT ID's.

closed 2018-05-04 by jettisonjoe