google/openhtf #638: Noisy tracebacks even though BlankDutIdError is caught

DanLipsitt, 2017-08-08:
Catching `BlankDutIdError` doesn't prevent tracebacks from being printed:

```python
import openhtf as htf

test = htf.Test()

try:
    test.execute()
except htf.core.test_state.BlankDutIdError:
    print "Caught BlankDutIdError."
```

```pytb
ERROR:root:Thread raised an exception: TestExecutorThread
Traceback (most recent call last):
  File "/Users/dan/.virtualenvs/tmp-2f36a8b19fca1ff5/lib/python2.7/site-packages/openhtf/util/threads.py", line 60, in run
    self._thread_proc()
  File "/Users/dan/.virtualenvs/tmp-2f36a8b19fca1ff5/lib/python2.7/site-packages/openhtf/core/test_executor.py", line 148, in _thread_proc
    self._execute_test_phases(phase_exec)
  File "/Users/dan/.virtualenvs/tmp-2f36a8b19fca1ff5/lib/python2.7/site-packages/openhtf/core/test_executor.py", line 187, in _execute_test_phases
    self.test_state.finalize_normally()
  File "/Users/dan/.virtualenvs/tmp-2f36a8b19fca1ff5/lib/python2.7/site-packages/openhtf/core/test_state.py", line 258, in finalize_normally
    self._finalize(test_record.Outcome.PASS)
  File "/Users/dan/.virtualenvs/tmp-2f36a8b19fca1ff5/lib/python2.7/site-packages/openhtf/core/test_state.py", line 293, in _finalize
    'Blank or missing DUT ID, HTF requires a non-blank ID.')
BlankDutIdError: Blank or missing DUT ID, HTF requires a non-blank ID.
Exception in thread TestExecutorThread:
Traceback (most recent call last):
  File "/usr/local/Cellar/python/2.7.13_1/Frameworks/Python.framework/Versions/2.7/lib/python2.7/threading.py", line 801, in __bootstrap_inner
    self.run()
  File "/Users/dan/.virtualenvs/tmp-2f36a8b19fca1ff5/lib/python2.7/site-packages/openhtf/util/threads.py", line 60, in run
    self._thread_proc()
  File "/Users/dan/.virtualenvs/tmp-2f36a8b19fca1ff5/lib/python2.7/site-packages/openhtf/core/test_executor.py", line 148, in _thread_proc
    self._execute_test_phases(phase_exec)
  File "/Users/dan/.virtualenvs/tmp-2f36a8b19fca1ff5/lib/python2.7/site-packages/openhtf/core/test_executor.py", line 187, in _execute_test_phases
    self.test_state.finalize_normally()
  File "/Users/dan/.virtualenvs/tmp-2f36a8b19fca1ff5/lib/python2.7/site-packages/openhtf/core/test_state.py", line 258, in finalize_normally
    self._finalize(test_record.Outcome.PASS)
  File "/Users/dan/.virtualenvs/tmp-2f36a8b19fca1ff5/lib/python2.7/site-packages/openhtf/core/test_state.py", line 293, in _finalize
    'Blank or missing DUT ID, HTF requires a non-blank ID.')
BlankDutIdError: Blank or missing DUT ID, HTF requires a non-blank ID.

Finishing test with outcome ABORTED.
Finishing test execution early due to test abortion, outcome ABORTED.
Caught BlankDutIdError.
```

fahhem, 2017-08-09:
I believe errors are normally printed 3 times due to some overzealous `logging.exception()` calls. Reducing this by 1 or to 1 is preferred behaviour, but it requires tracking down every place that logs and choosing which one covers the most cases and which ones are redundant

DanLipsitt, 2017-08-10:
I don't understand why tracebacks are being printed inside the library at all. To me it seems like fatal exceptions should bubble up to the `__main__` level and be handled (or not) by the application script, and non-fatal exceptions should be handled and logged in a user-friendly way.

DanLipsitt, 2017-08-10:
I guess part of the problem is that we are using the console for user interaction right now (because we can't get the GUI to work for us). So the workaround would be to configure our loggers differently. But I still don't want redundant tracebacks in our logs if we can help it.

fahhem, 2017-08-10:
The first problem is that we're logging the same exceptions in multiple places, which I agree should be fixed.

The next is fatal exceptions that are caught, logged, and swallowed by OpenHTF should instead be allowed to raise out of `.execute()`. I'm not sure I'm on board with that one. The approach I have is that `.execute()` should run everything and call the output callbacks and return `True`/`False` based on pass/failure. A lower-level API, one that I'm designing (mostly in my head at the moment), would expose more of this sort of stuff but the existing 'high-level API' would pass it all into the output callbacks and should never raise an exception.

jettisonjoe, 2018-05-05:
This was fixed as part of the recent logging refactor.

closed 2018-05-05 by jettisonjoe