google/openhtf #644: How to handle errors during plug init

DanLipsitt, 2017-08-23:
What's the correct way to handle exceptions that occur during plug init? For example, if a serial port isn't found because a device isn't connected.

fahhem, 2017-08-24:
At the moment, OpenHTF doesn't handle this well at all. I think we get a traceback and exit without a DUT ID so the output callbacks aren't called (as you've said in #636).

I don't have any good ideas here, other than calling output callbacks with the test outcome being ERROR, though we don't even store plug init or teardown information anywhere (such as how long it took, errors raised during them, etc). It would be good to add that all

DanLipsitt, 2017-08-24:
OK, thanks.
Related question: Is it feasible to prompt the user during plug init? "Please make sure the usb cable is plugged in." That sort of thing. Calling one plug from another feels risky.

fahhem, 2017-08-24:
Plug dependencies is actually something I've wanted to add (back) into OpenHTF, but it's not there at the moment. One team here implemented that themselves, and I think they might share it externally if you want it (@arsharma1 wrote it). You can use `ConsolePrompt` from any plug, since it's not a plug and it handles cross-platform prompts for you. If you want to use `UserInput` for its frontend features, you can't from another plug since it won't be the instance the framework created and tracks for frontend updates.

arsharma1, 2017-08-24:
I'll add it to contrib with a pull request.

It does not currently work with Placeholders, though.

They get declared with descriptors and require iterating over all the phases:
```
class NeededPlug(plug.BasePlug):
  pass

class AnotherPlug(plugs.BasePlug):
  needed = PlugDependency(NeededPlug)

  def action(self):
    # use self.needed as required.

def main():
  phases = [phase1, phase2, phase3]
  init_phase = create_dependency_init_phase(phases)
  phases.insert(0, init_phase)
  test = htf.Test(phases, ...)
```

arsharma1, 2017-08-24:
Please see https://github.com/google/openhtf/pull/645