google/openhtf #647: Test record logging problems

DanLipsitt, 2017-09-05:
We want to be able to 
* log to the console through a handler of our choice ([colorlog](https://github.com/borntyping/python-colorlog))
* log to `log_records` in our test output
* set test record loglevels to INFO or DEBUG as needed
* not have duplicate logging to the console

I'm willing to hack up `openhtf.logs.util` to get it done, but I can't for the life of me seem to get all of these things to happen at once. Console logging and record output logging seem to be tied together, but I can't tell where. 

I've seen the note that says "Test record logs are by default output to stdout at a debug level.  There is no way to change this." Why?

DanLipsitt, 2017-10-03:
Some things that make logging configuration hard:
* `propagate` is set to `False` in the main logger: https://github.com/google/openhtf/blob/6e05dbedb28f8a2739fd565407457f606de91741/openhtf/util/logs.py#L203
* The `record_logger` deletes existing handlers: https://github.com/google/openhtf/blob/6e05dbedb28f8a2739fd565407457f606de91741/openhtf/util/logs.py#L120-L121

Note that it's probably necessary to be able to add handlers before `Test`s are created or `Plug` initializers (for example) won't be able to log correctly.

jettisonjoe, 2018-05-05:
Now all handlers are on the top-level 'openhtf' logger, so you should be able to go in and change them as long as you do it after you call `configure_cli_logging()`.

closed 2018-05-05 by jettisonjoe