google/openhtf #649: Block signals during a phase

DanLipsitt, 2017-09-06:
We have a phase that writes system flash that we want to guard from interruption. Since tests and plugs don't run on the main thread and `signal.signal()` has to be called from the main thread, I can't use it. Can't catch `KeyboardInterrupt` because the main thread's handler intercepts it. Any suggestions?

fahhem, 2017-09-06:
Your code runs openhtf.Test().execute() can disable signals. Alternatively,
you can run execute() in a separate thread and have your main thread handle
listening to a queue of events for enabling/disabling signals if you want
to only disable them during that phase and no others.

Does that seem reasonable? I'm not sure if there's a good way to have
OpenHTF do it for you, since we're not guaranteed to run from the main
thread either (such as in the second suggestion)

DanLipsitt, 2017-09-06:
Yes, a queue sounds doable, if slightly complicated (but necessarily so).

If I want to disable signals, I think it needs to come between test creation and .exectue()? Otherwise it will be overwritten by OpenHTF.

closed 2022-10-13 by glados-verma