google/openhtf #657: Add ability to mock out time.sleep in phase unit tests

wallacbe, 2017-10-11:
Add ability or add example how to mock out time.sleep in phase unit tests

fahhem, 2017-10-11:
Can you just mock it out yourself using mock.patch.object(time, 'sleep')?

wallacbe, 2017-10-11:
I tried, that.  Unsure if it's an issue with my syntax, or some interaction with mocking plugs at the same time?

    class HelloWorldTest(test.TestCase):
  File "all_the_things_test.py", line 19, in HelloWorldTest
    @mock.patch.object(time, 'sleep')
  File "/home/source/venv/local/lib/python2.7/site-packages/openhtf/util/test.py", line 283, in test_wrapper
    'it.' % (mock_name, test_func.__name__), plug_args)
openhtf.util.test.InvalidTestError: ('patch_plugs got kwarg mock_prompt, but test method testPass does not expect it.', set([]))

`import time
import unittest
import all_the_things
import mock
import openhtf
from openhtf.util import test

MOCK_PLUG = 'example_plugs.ExamplePlug'
MOCK_PROMPT = 'openhtf.plugs.user_input.UserInput'

class HelloWorldTest(test.TestCase):

  def setUp(self):
    self.phase = all_the_things.hello_world

  @test.patch_plugs(mock_prompt=MOCK_PROMPT, mock_plug=MOCK_PLUG)
  @mock.patch.object(time, 'sleep')
  def testPass(self, mock_sleep, mock_prompt, mock_plug):
    test_rec = yield openhtf.Test(self.phase)
    self.assertTestFail(test_rec)

if __name__ == '__main__':
  unittest.main()
`

fahhem, 2017-10-11:
What if you changed the order? I think `test.patch_plugs` might not be happy with how `mock.patch.object` does things?

wallacbe, 2017-10-11:
I changed the order and saw a different error:

  class HelloWorldTest(test.TestCase):
  File "all_the_things_test.py", line 19, in HelloWorldTest
    @test.patch_plugs(mock_prompt=MOCK_PROMPT, mock_plug=MOCK_PLUG)
  File "/home/source/venv/local/lib/python2.7/site-packages/openhtf/util/test.py", line 278, in test_wrapper
    'patch_plugs kwargs: ' % (test_func.__name__, arg), mock_plugs)
openhtf.util.test.InvalidTestError: ('Test method testPass expected arg mock_sleep, but it was not provided in patch_plugs kwargs: ', {'mock_prompt': 'openhtf.plugs.user_input.UserInput', 'mock_plug': 'example_plugs.ExamplePlug'})

```
import time
import unittest
import all_the_things
import mock
import openhtf
from openhtf.util import test

MOCK_PLUG = 'example_plugs.ExamplePlug'
MOCK_PROMPT = 'openhtf.plugs.user_input.UserInput'

class HelloWorldTest(test.TestCase):

  def setUp(self):
    self.phase = all_the_things.hello_world

  @mock.patch.object(time, 'sleep')
  @test.patch_plugs(mock_prompt=MOCK_PROMPT, mock_plug=MOCK_PLUG)
  def testPass(self, mock_prompt, mock_plug, mock_sleep):
    test_rec = yield openhtf.Test(self.phase)
    self.assertTestFail(test_rec)

if __name__ == '__main__':
  unittest.main()
```

wallacbe, 2017-10-12:
Mocking sleep with a with statement appears to work:

```
import time
import unittest
import all_the_things
import mock
import openhtf
from openhtf.util import test

MOCK_PLUG = 'example_plugs.ExamplePlug'
MOCK_PROMPT = 'openhtf.plugs.user_input.UserInput'

class HelloWorldTest(test.TestCase):

  def setUp(self):
    self.phase = all_the_things.hello_world

  @test.patch_plugs(mock_prompt=MOCK_PROMPT, mock_plug=MOCK_PLUG)
  def testPass(self, mock_prompt, mock_plug):
    with mock.patch.object(time, 'sleep'):
      test_rec = yield openhtf.Test(self.phase)
      self.assertTestFail(test_rec)

if __name__ == '__main__':
  unittest.main()
```

closed 2017-10-25 by wallacbe