google/openhtf #665: Exception that results in test FAIL

kdsudac, 2017-10-27:
Current behavior is any uncaught exception causes the test to be classified as an ERROR.  In some cases the h/w or h/w driver raises the exception because of a detected h/w failure.  

To properly classify as an error, the test author has to catch the exception and return PhaseResult.STOP.  However, this approach is a bit clumsy because it is likely to require try/catch in many phases (e.g. if a h/w driver is used in almost every phase of a test).

Proposed solutions:

1)  Create a base exception TestFailure that test authors can subclass.  Modify PhaseExecutorThread to catch any subclass of this exception and classify as a FAIL.
2)  Allow test author to specify test-level option of list of Exceptions to classify as an FAIL.  i.e. @htf.TestOptions(failure_exceptions=[NetworkError, LowVoltageError])

fahhem, 2017-10-27:
cpaulin should be working on something along the lines of explanation
callbacks that would provide this capability with a lot more power. In
order to enable its full power, I think `PhaseDescriptor` should provide a
way for decorators that aren't crucial to the framework to add data to a
phase, so adding a "extra" or "misc" dict attribute would be good.

That would allow a decorator along the lines of
`@FailureExceptions(NetworkError, LowVoltageError, reason='...')` to then
shove that data into `extra` and be understood by an explanation callback
that gets a `TestRecord` where the last Phase had a terminal result and its
exception data was in `extra['failure_exceptions']`.

To clarify explanation callbacks, they're similar to output callbacks in
their guarantees (always called after a test started, even if plugs
completely failed to be created) but they have write-access to the
`TestRecord` and run before the output callbacks. This allows callbacks to
"explain" the test result, such as converting known bad measurements into
operator- or MES-friendly outcome details like "BadDiodeD12" or
"MissingPowerSupply"

wallacbe, 2017-10-27:
This old issue seems related:
https://github.com/google/openhtf/issues/522
"Add need pass property for measurements"
Add ability for an out of spec measurement to return PhaseResult.STOP.

kdsudac, 2017-10-27:
Note: should also add outcome detail with the exception's name.

kdsudac, 2017-10-27:
Talked to kschiller, we think my proposed solution #2 is best approach (unless cpaulin is actively working on explanation callbacks).  I'm willing to work on this feature unless people have strong objections.

We aren't sure if it would make sense to apply at the Test or Phase level.  Test level may not be granular, It can be cumbersome to have to apply the same decorator to 15 different phases to catch the same failure.

glados-verma, 2022-10-13:
Not planned for openhtf.

Proposal 1 would get limited adoption as this would require plug writers to include openhtf libraries.
Proposal 2 is easy but it violates the principle of make a try/except block as small as possible. You could easily end up catching exceptions raised from parts of the code you didn't expect, and suddenly an operator is trying a troubleshooting step that's not applicable. Instead users are advised to use narrowly scoped try/except blocks in their phases.

blaisegassend, 2022-10-13:
Is there a way of achieving this goal that doesn't involve including openhtf libraries?

I didn't understand the objection to proposal 2 since it includes an explicit list of exceptions to catch. Sure, the framework would have to catch all exceptions, but it would then reraise all the non-relevant ones.

dbhatman, 2022-10-13:
Proposal 2 already exists and is implemented: #698 
The framework has already been catching all exceptions and analyzing them anyway.

For @blaisegassend and @glados-verma there are a few examples of users using this internally. Just search for failure_exceptions internally or ping me directly and I can point this to you.

closed 2022-10-13 by dbhatman

glados-verma, 2022-10-13:
Interesting, didn't realize this was already implemented :) The caution I mentioned above still applies in its usage; the exception could be coming from any phase of your test, and could easily be from a part of the code you didn't expect. E.g. you might declare an unreachable device exception to be a failure; a fixture fault develops over time that causes this exception to come from pre-test portions of the code, and suddenly a bunch of runs that should be errors are instead failures. Something to keep in mind if you use this pattern.