google/openhtf #687: How can I allow an operator to choose which phases to run?

shaperilio, 2018-01-05:
I need to run a test where the operator chooses the phases at the start. Ideally, I wouldn't have to go and change any phase code, so I can quickly change the list of phases available to the operator. Ideally, I'd somehow alter the phase list and update the `run_if` option after the operator has chosen which phase he wants to run.

Is this possible?

fahhem, 2018-01-12:
What was done internally is have the phase selection happen before the OpenHTF test was created, then we simply pass a different list of phases to the `Test` to execute it. Would that work for you?

shaperilio, 2018-01-12:
Yes, that's what I ended up doing.

Basically we have several tests that run as standard procedure. Then we want to have one place where any phase from any test can be run on an as-needed basis. So the ideal flow would be:

1. Show menu
2. Operator chooses a phase.
3. Phase executes.
4. Show menu
5. Operator can then end the test, repeat the last phase, or select a new one.

I think this behavior could only be done from within a test, e.g. "adding phases on the fly".

Because I now have to create the test a-priori, repeating a phase would mean running a new test, which adds a bit of annoyance to the operator (as we have a few initialization phases he must go through).