google/openhtf #688: Consistency of dictionary iteration behavior between Python 2 and 3

Kenadia, 2018-01-06:
#679 introduced support for Python 3, in many cases converting dict iteration code, e.g. `d.itervalues()` -> `d.values()`. In many cases, we are left with code that has different semantics in Python 2 and 3, e.g. `d.values()` produces a list in Python 2, and a view object in Python 2.

This doesn't currently appear to be a problem, but down the line, so long as we maintain OpenHTF as a hybrid Python 2/3 project, this may make things harder for maintainers, who will have to consider different behaviors between the two versions.

[PEP 469](https://www.python.org/dev/peps/pep-0469/) discusses this problem and proposes solutions. For example, helper functions can be used to succinctly convert Python 2 code to hybrid code:
```
d.keys() -> list(d)
d.values() -> listvalues(d)
d.items() -> listitems(d)
d.iterkeys() -> iter(d)
d.itervalues() -> itervalues(d)
d.iteritems() -> iteritems(d)
```

fahhem, 2018-01-06:
six is a single file library that provides these helpers and many others.
If we don't want to add it as a dependency, then we can include it inside
openhtf. As long as we update for security patches then it's safe/okay.

Kenadia, 2018-03-22:
Partially addressed by #737

closed 2022-10-13 by glados-verma