google/openhtf #704: web gui does not register clients

arnew, 2018-01-22:
Regardless of running the web_gui in a virtualenv of using a directly, upon running the hello_world.py example (locally, or on another machine), the web_gui emits an error.

```
$ python -m openhtf.output.web_gui
Starting openhtf web gui server on http://localhost:12000.
Exception in thread Thread-1:
Traceback (most recent call last):
  File "/usr/lib/python2.7/threading.py", line 801, in __bootstrap_inner
    self.run()
  File "/usr/local/lib/python2.7/dist-packages/openhtf/output/web_gui/__init__.py", line 247, in run
    self._handle_stations(iter(self.stations.keys()))
  File "/usr/local/lib/python2.7/dist-packages/openhtf/output/web_gui/__init__.py", line 231, in _handle_stations
    self.watch_tests(hostport)
  File "/usr/local/lib/python2.7/dist-packages/openhtf/output/web_gui/__init__.py", line 259, in watch_tests
    hostport, remote_test, self._on_update_callback)
  File "/usr/local/lib/python2.7/dist-packages/openhtf/output/web_gui/__init__.py", line 139, in __init__
    self.handle_state_update()
  File "/usr/local/lib/python2.7/dist-packages/openhtf/util/threads.py", line 138, in synchronized_method
    return func(self, *args, **kwargs)
  File "/usr/local/lib/python2.7/dist-packages/openhtf/output/web_gui/__init__.py", line 153, in handle_state_update
    state = self._test.state
  File "/usr/local/lib/python2.7/dist-packages/openhtf/core/station_api.py", line 124, in _wrapper
    return func(*args, **kwargs)
  File "/usr/local/lib/python2.7/dist-packages/openhtf/core/station_api.py", line 435, in state
    self.test_uid, summary_dict and summary_dict['test_record'])
  File "/usr/local/lib/python2.7/dist-packages/openhtf/util/xmlrpcutil.py", line 98, in _wrapper
    return method(*args)
  File "/usr/lib/python2.7/xmlrpclib.py", line 1243, in __call__
    return self.__send(self.__name, args)
  File "/usr/lib/python2.7/xmlrpclib.py", line 1602, in __request
    verbose=self.__verbose
  File "/usr/lib/python2.7/xmlrpclib.py", line 1283, in request
    return self.single_request(host, handler, request_body, verbose)
  File "/usr/lib/python2.7/xmlrpclib.py", line 1316, in single_request
    return self.parse_response(response)
  File "/usr/lib/python2.7/xmlrpclib.py", line 1493, in parse_response
    return u.close()
  File "/usr/lib/python2.7/xmlrpclib.py", line 800, in close
    raise Fault(**self._stack[0])
Fault: <Fault 1: "<type 'exceptions.TypeError'>:cannot marshal <class 'openhtf.plugs.PlugDescriptor'> objects">

^CERROR:__main__:Received SIGINT. Stopping web server.
```

pupssman, 2018-01-30:
Ran into same issue. Any ideas / progress?

Kenadia, 2018-02-08:
@arnew @pupssman Sorry for the slow progress here. This should be fixed by #721. The old web GUI does not get a lot of attention these days. It will be replaced by the brand new GUI in #720.

closed 2018-02-23 by Kenadia