google/openhtf #765: Frontend Import Issue with Guzzle

marisdu, 2018-04-27:
I am trying to run the example frontend_example.py script in my virtual environment and I am getting this error: 
`Traceback (most recent call last):
  File "frontend_example.py", line 21, in <module>
    from openhtf.output.servers import station_server
  File "/Users/yuka/Desktop/Work/Treeline/Boom/avmsa_anchor/src/open_htf/openhtf/openhtf/output/servers/station_server.py", line 26, in <module>
    from openhtf.output.callbacks import mfg_inspector
  File "/Users/yuka/Desktop/Work/Treeline/Boom/avmsa_anchor/src/open_htf/openhtf/openhtf/output/callbacks/mfg_inspector.py", line 48, in <module>
    from openhtf.output.proto import guzzle_pb2 
ImportError: cannot import name guzzle_pb2`

I am able to successfully run scripts not dependent on the GUI.

marisdu, 2018-04-30:
SOLVED: Was an issue with compiling using protoc! 

great resource: https://developers.google.com/protocol-buffers/docs/pythontutorial

closed 2018-04-30 by marisdu

pizazzyyy, 2018-05-04:
Got it. We need to compile Protocol Buffers using "protoc -I=$SRC_DIR --python_out=$DST_DIR $SRC_DIR/guzzle.proto". Thanks for the resource link!

aospan, 2018-08-18:
quick fix here:

```
export PROTO_DIR=openhtf/output/proto/
protoc -I=$PROTO_DIR --python_out=$PROTO_DIR $PROTO_DIR/guzzle.proto
protoc -I=$PROTO_DIR --python_out=$PROTO_DIR $PROTO_DIR/test_runs.proto
```

should it be done in ```python setup.py install``` ?

aospan, 2018-08-18:
Sorry, my bad. Found in readme:

Note: some of the examples/ require protocol buffer code to be generated via python setup.py build command.

lucabiason, 2020-05-05:
I am sorry is not clear to me. What I am supposed to do when created the .proto file you are suggesting 

thanks

lucabiason, 2020-05-07:
Sorry, 
once you solved the guzzle it appears test_runs.proto errors

test_runs.proto:26:8: "google.protobuf.EnumValueOptions" is not defined.
test_runs.proto: "google.protobuf.EnumValueOptions" is not defined.
test_runs.proto:23:1: warning: Import descriptor.proto is unused.

Do you have any idea how to move forward?