google/openhtf #766: Allow Monitors to use Validators

andrewwade, 2018-04-27:
I'd like to be able to validate monitored values as they are read. I don't think this is possible because the `monitors()` function doesn't allow you to pass a `Measurement` object.

Example:
```
@monitors(Measurement('current').in_range(0.2, 0.5), poll_Interval_ms=50)
def turn_on(test):
    do_turn_on()
    wait_for_init(seconds=1)
```

Straeng, 2020-10-29:
Hi,
Is something planned for this issue? 

I have a scenario with a long-running test case in which I want to monitor a temperature to make sure it does not exceed a given value. 

What is the preferred way of doing this? I can't use a dimensioned measurement since I want to abort the test immediately if the temperature rises above the limit. It sounds like adding a validator to a monitor would be one soultion.

cricdecyan, 2020-10-29:
Hi,
You could have another phase that validates the measurements of your
monitored phase.
Another idea is to modify monitors.py to use DimensionPivot from
openhtf/utils to validate the multidims
Regards
Christian