google/openhtf #790: OpenHTF hangs after positing to database, but before completing.

sciguy14, 2018-05-29:
We observe an issue roughly 10% of the time where an OpenHTF test completes but then hangs and never returns to the parent thread. It seems like one of the threads in OpenHTF is hanging at exit, but we haven't been able to figure out what thread it is. Have you folks seen this issue before? Any idea what steps we should take to try to figure out why this is happening?

If you hit Ctrl-C to kill it once it is hung, this is the output:
[image]

We're not sure if it's related, but we've also seen issues with tests starting in an already stopped state. It also seems to be some sort of threading race condition. We were only able to resolve it by adding a 1 second sleep into __init__.py after `self._executor.start()`: https://github.com/ShaperTools/openhtf/commit/ccdc01494a4c67944489cfcc873207b1862cbac5

cc: @shaperilio

jettisonjoe, 2018-06-07:
I haven't seen OpenHTF do a silent hang like that in a super long time, and the times I have seen it it has almost always been due to a plug getting blocked on I/O. The executor race condition is interesting as well. It may or may not be related. We should definitely investigate.

sciguy14, 2018-06-08:
@jettisonjoe is there any additional info that I can provide to help with investigation? Happy to help.

Kenadia, 2018-06-15:
Regarding the race condition, that sounds like some of the things @arsharma1 has been working on recently.

The thread hanging issue sounds more use-case dependent to me, we might need more context to investigate. On my team we've used a little snippet like below, running after `test.execute()` to debug issues with hanging threads. We try to maintain a habit of passing a `name` argument to new threads to help with this kind of debugging.

```
  daemon_threads = lambda: sum(t.daemon for t in threading.enumerate())
  while threading.active_count() != 1 + daemon_threads():
    for thread in threading.enumerate():
      print 'Living thread:', thread.name, thread.daemon
    time.sleep(5)
```

(On the other hand if this is an issue with `text.execute()` not returning, then I would of course start by checking if all the output callbacks are finishing successfully.)

fahhem, 2018-06-21:
I would also print out whether each thread is daemon or not (`thread.daemon`). Only non-daemon threads can block

Kenadia, 2018-06-25:
Yes, I would print out `thread.daemon`. Also note that in our version (originally authored by @fahhem) we use `while threading.active_count() != 1 + daemon_threads()` to determine whether to exit the loop, instead of relying on a magic number of threads.

@shaperilio, in your five threads, besides the main thread, it appears that two are ThreadPool threads from `requests_futures.sessions` and the other two are `openhtf.util.multicast.MulticastListener` and `openhtf.core.station_api.ApiServer`. All of these should be daemon threads, so I'm not sure what could cause it to hang. I would look into why the ThreadPool threads are there, in case they shouldn't be. Otherwise, this may somehow be related to multicast and station_api—however, station_api.py was deleted in OpenHTF, and replaced with station_server.py and a new GUI in #727.

Please let us know if the issue reappears or you learn more about the cause. Happy to help if we can, but I propose closing this issue unless there's evidence that it affects OpenHTF at head.

shaperilio, 2018-06-27:
@Kenadia I'll keep you guys posted; your code is in there, but we haven't seen it hang in months.

I last updated OpenHTF on March 26th when we were getting a lot of hangs and there was no change then.

closed 2018-06-27 by Kenadia