google/openhtf #791: UNKNOWN_DUT test record is output when sig-int is sent at start prompt

wallacbe, 2018-06-09:
See the following files output when running all the things and doing sig int at the prompt.  Expectation would be to not get a test record output correct?

UNKNOWN_DUT.MyTest.1528518533178.pickle
UNKNOWN_DUT.MyTest.1528518533178.json 

~/openhtf/examples$ python all_the_things.py 
Enter a DUT ID in order to start the test.
--> ^C
============== test: MyTest  outcome: ABORTED ==============

Traceback (most recent call last):
  File "all_the_things.py", line 172, in <module>
    test.execute(test_start=user_input.prompt_for_test_start())
  File "/home/venv/local/lib/python2.7/site-packages/openhtf-1.3.0-py2.7.egg/openhtf/core/test_descriptor.py", line 293, in execute
    self._executor.wait()
  File "/home/venv/local/lib/python2.7/site-packages/openhtf-1.3.0-py2.7.egg/openhtf/core/test_executor.py", line 109, in wait
    self.join(31557600)
  File "/usr/lib/python2.7/threading.py", line 951, in join
    self.__block.wait(delay)
  File "/usr/lib/python2.7/threading.py", line 359, in wait
    _sleep(delay)
  File "/home/venv/local/lib/python2.7/site-packages/openhtf-1.3.0-py2.7.egg/openhtf/core/test_descriptor.py", line 234, in handle_sig_int
    raise KeyboardInterrupt()
KeyboardInterrupt

Kenadia, 2018-06-11:
The default_dut_id feature was tacked onto #756. When we added the feature we didn't consider that aborts at test start should be handled differently. So perhaps the expected behavior is that output callbacks should not run if test start doesn't complete?

I know @arsharma1 has some test executor changes in progress relating to better support for teardown phases, and we discussed the need for a design doc for the executor behavior. I think we should take the time to define the expected behavior of the executor from start to finish, and consider this issue then.

wallacbe, 2018-06-11:
Did Arun want to log unexpected aborts during initialization during start
up phase?  My abort case would be an expected abort to stop the tester at
the start prompt.

jettisonjoe, 2018-06-11:
Ah, yeah, this behavior change is sort of an accident, but is arguably "more correct" than the old behavior. Before there would be no test record, but only because the output module would blow up when trying to access the non-existent DUT ID. I didn't anticipate this case when I authored the change. I'm not sure what we intend/prefer.

arsharma1, 2018-06-12:
Not outputting when the error occurs in the trigger phase sounds good to me.  I just want to make sure we get logs if plug initialization fails.  I'm not sure how that would work if the engine errors out when initializing plugs for the trigger phase, though.

arsharma1, 2018-06-12:
The phase and Test Executor don't see the KeyboardInterrupt, they just get notified via the abort logic.  That sounds good, though.

Kenadia, 2018-06-12:
My only concern is that by singling out explicit aborts in the test start only, we're making the executor behavior unnecessarily complicated and opinionated. Thus far, the executor doesn't really distinguish between aborts and other errors. But if every team agrees this is just common sense, then it seems that's the way to go.

For what it's worth, here is what the workaround looks like if a team had to implement this at the callback level:

```python
class CustomOutputCallback(callbacks.OutputToFile):

  def __call__(self, test_record):
    if not (test_record.outcome == test_record.Outcome.ABORTED
            and test_record.dut_id == 'UNKNOWN_DUT')
      super(CustomOutputCallback, self).__call__(test_record)
```

jettisonjoe, 2018-06-12:
@Kenadia I think that's a good instinct (avoiding complexity and strong opinions in the framework), and the test author driven workaround is small and simple. Maybe we could put an example like this into the examples/ directory and explain what it's for and that might be good enough?