google/openhtf #802: KillableThread cannot be killed when in time.sleep

arsharma1, 2018-06-19:
```python
import time
import threading

def unkillable_phase(test):
  print('hello start')
  time.sleep(1)
  print('hello middle')
  time.sleep(2)
  print('hello end')

def alternate_killable_phase(test):
  print('hello start')
  threading.Event().wait(1)
  print('hello middle')
  threading.Event().wait(2)
  print('hello end')
```

The first phase cannot be killed with CTRL-C, but the second one can be.

fahhem, 2018-06-19:
Yeah, this is because of the way the default `time.sleep()` is implemented (a C-level `select` with the GIL unlocked) versus a `Event.wait()` (based on `Condition.wait()`, which does a Python-level loop around a non-blocking acquire + sleep). The Ctrl-C works in the Event case because the sleep is at most 50ms.

Not a great solution here, but one thing that was done way back in the day, was to replace the builtin time.sleep with the equivalent of `threading.Event().wait`. However, that would only affect PY2, as I just checked the implementation of PY3's `Event.wait()` and it's a call into C in the end because PY3 has a timeout on their locks (as of 3.2) and Condition.wait is now based on a C level call into that function.

KillableThread requires Python's interpreter to check for exceptions and throw that exception in that thread, if the interpreter isn't running (because the GIL has been released while sleeping), then KillableThread has no power. The only way to really do this is to super-duper cheat and call into `pthread_kill()` somehow, but again that'll only work on Linux.

One thing I had been thinking of that would solve this and a few other issues, is to somehow split the test into a control process and 1 or more runner processes. There would be a runner per test (and one thread in the control process per test) and that runner would own the plugs, logs, etc (helping us split the logs to only the relevant tests, for instance), and the runner would use an `os.pipe` to push data back to the control process's thread. If that process ever raised an exception, it could push that data and exit, cleaning up all plugs in the process. If it was being cancelled, we'd just use the normal `Process.kill()` and `.terminate()` functions. If it survived the run, we could reuse the plugs (or kill and restart if wanted). 

It would be a pretty fundamental change though, so I've considered it essentially as an openhtf2 or 2.0 sort of thing.

TL:DR; KillableThread is and always has been a hack. If you really want to kill a running test, it's gotta be a process.

jettisonjoe, 2018-06-19:
For the record, I'm loosely in favor of designing a separate-process-per-execute() model for v2.0. In addition to helping with test interruption, it would also guard against certain kinds of memory leaks and accidental state bleedover from one test run to the next.