google/openhtf #833: Access arguments passed to phases from monitors

alireza-hosseini, 2018-11-27:
I'm trying to access the argument (`node_id`) I provided to my phase from a monitor and here is my code:
```
@htf.plug(mmp=motion_master_plug.MotionMasterPlug)
def speed_monitor(test, mmp, node_id):
    return mmp.get_velocity(node_id)

@htf.monitors("velocity", speed_monitor)
@htf.plug(mmp=motion_master_plug.MotionMasterPlug)
def test_velocity_control(test, node_id):
    time.sleep(5)
    # Do something here

if __name__ == '__main__':
    test = htf.Test(test_velocity_control.with_args(node_id=0))
    test.execute(test_start=lambda: "test")
```

But the test execution terminates will the following error:

> Error: __call__() got an unexpected keyword argument 'node_id'

This error only appears if I want to use the monitor and it is completely fine with the phase itself.

cricdecyan, 2018-11-27:
Hi Alireza,
How about adding mmp to the declaration of test_velocity? Right now it is
missing.
Regards
C

alireza-hosseini, 2018-11-27:
@cricdecyan You're right, it was missing but adding it doesn't help to resolve the main issue.

jettisonjoe, 2018-11-28:
Can you run the test with -v so we can see the full stack trace?

alireza-hosseini, 2018-11-29:
@jettisonjoe Here is the full stack trace:
```C 10:09:11 threads - Thread raised an exception: velocity_MonitorThread
C 10:09:11 <phase: speed_monitor> - Phase test_velocity_control raised an exception
C 10:09:11 <phase: speed_monitor> - Finishing test execution early due to an exception raised during phase execution; outcome ERROR.
C 10:09:11 <phase: speed_monitor> - Traceback:
  File "/home/alireza/workspace/test_htf/venv/local/lib/python2.7/site-packages/openhtf/util/threads.py", line 60, in run
    self._thread_proc()
  File "/home/alireza/workspace/test_htf/venv/local/lib/python2.7/site-packages/openhtf/core/phase_executor.py", line 155, in _thread_proc
    phase_return = self._phase_desc(self._test_state)
  File "/home/alireza/workspace/test_htf/venv/local/lib/python2.7/site-packages/openhtf/core/phase_descriptor.py", line 243, in __call__
    **kwargs)
  File "/home/alireza/workspace/test_htf/venv/local/lib/python2.7/site-packages/openhtf/core/monitors.py", line 148, in monitored_phase_func
    return phase_desc(test_state, *args, **kwargs)

:ERROR
  exception type: TypeError

E 10:09:11 test_descriptor - Output callback <openhtf.output.callbacks.json_factory.OutputToJSON object at 0x7f708e4eb450> raised; continuing anyway

```

jettisonjoe, 2018-12-04:
The way the phase and monitor functions are stored inside the framework, they each have their own "PhaseDescriptor" wrapper. Calling .with_args() on the phase itself's descriptor doesn't add the same args to the associated monitor descriptor.

My suggestion is to add another phase, maybe a set_node_id_phase or similar, and have it set the node_id in test state before the monitored phase is run:

```python
set_node_id_phase(test, node_id):
  test.state['node_id'] = node_id

@htf.plug(mmp=motion_master_plug.MotionMasterPlug)
def speed_monitor(test, mmp):
    return mmp.get_velocity(test.state['node_id'])

@htf.monitors("velocity", speed_monitor)
@htf.plug(mmp=motion_master_plug.MotionMasterPlug)
def test_velocity_control(test, mmp):
    time.sleep(5)
    # Do something here

if __name__ == '__main__':
    test = htf.Test(
      set_node_id_phase.with_args(node_id=0),
      test_velocity_control)
    test.execute(test_start=lambda: "test")
```

arsharma1, 2018-12-04:
I've not been satisfied with the phase wrapping implementation in core/monitors.py.  I have a more extensive mechanism for having phases calling a nested phase that is compatible with with_plugs and with_args.  I will work on upstreaming it.