google/openhtf #836: Start test based on input from serial port

elliotmakuh, 2018-12-04:
I'm new to using OpenHTF but would like to start my test when a certain packet is received from a serial port.  I have been trying to wrap my head around the Prompts (which is what I think this would end up being) but would appreciate some guidance on how to implement this functionality.

arsharma1, 2018-12-04:
Prompts are for user input.

In this case, you just want a Plug, which just wraps an interface.  The Plug here would wrap access to your serial port.  The OpenHTF engine creates a single instance of each Plug class registered to your Phases and shares that instance to the Phases when they are run.

```import openhtf
from openhtf import plugs

class MySerialPlug(plugs.BasePlug):
  def __init__(self):
    self.serial = <create serial object>

  def wait_for_start_packet(self):
    <add code to wait for the expected packet>

@openhtf.plug(my_serial=MySerialPlug)
def wait_for_start_packet(test, my_serial):
  my_serial.wait_for_start_packet()
  test.logger.info('Found the start packet')```

elliotmakuh, 2018-12-05:
Thank you so much for the feedback!

closed 2018-12-06 by elliotmakuh