google/openhtf #839: Attempting to decorate @htf.measures()

zamora18, 2018-12-11:
Hello, 
I am trying to make my own decorator to wrap the `@htf.measures()` decorator. The purpose of this is to have a function that creates several measurements (potentially loaded from different configuration files), instead of having several lines stacked on my phase.

This has been my first application with using decorators so forgive me if there are some mistakes. This is what I have tried:
```python
def my_measurements_decorator(func):
    """
    Decorator function to wrap @htf.measures. Creates the measurement 
    names and limits for DUT voltage signals.

    :param func: Decorated function
    :return: OpenHTF measurements
    """

    def wrapper_my_measurements_decorator(*args, **kwargs):
        # conf.signals_csv_filename can be updated (using conf.load_from_file) so that 
        # different voltage signals can be recorded depending on the test being run.
        signals_list_of_dicts = list(csv.DictReader(open(conf.signals_csv_filename)))
        new_measurements_list = []
        for sub_dict in signals_list_of_dicts:
            m_name = sub_dict['Signal_Name']  # One of the signal names is 'V_3P3'
            m_min = sub_dict['Min_Value']
            m_max = sub_dict['Max_Value']
            m_doc = sub_dict['Info']
            m = htf.Measurement(m_name).in_range(m_min, m_max).doc(m_doc).with_units('volt')
            new_measurements_list.append(m)
        func(*args, **kwargs)
        return htf.measures(*new_measurements_list)
    return wrapper_my_measurements_decorator

@define_voltage_measurements
def check_voltages_phase(test):
    """
    Records each of the voltage signals of the DUT.
    """
    signals_list_of_dicts = list(csv.DictReader(open(conf.signals_csv_filename)))
    # Loop through each of the signals and record each voltage level as a test measurement
    for sub_dict in signals_list_of_dicts:
        signal_level = get_voltage_level(sub_dict['Signal_Name'])
        test.measurements[sub_dict['Signal_Name']] = signal_level  # Throws an error here
```

However when I run this I get this error:
```
:ERROR
  exception type: NotAMeasurementError

Error: ('Not a measurement', 'V_3P3', OrderedDict())
```
Which shows that none of the measurements are actually being added to the OpenHTF `test` object.

Am wrong in thinking that `wrapper_my_measurements_decorator` should return the `htf.measures()` decorator?

I've tried running without executing `func(*args, **kwargs)` but the obviously the phase will not execute.
I've also tried ending the wrapper with this:
```
    htf.measures(*measurements_list)
    func(*args, **kwargs)
return wrapper_my_measurements_decorator
```
(with and without a return statement before `func(*args, **kwargs)`), but get the same error as before.

I would appreciate any help that you can offer. Again, it's possible there may be some mistakes in the code/structure and I would also appreciate the feedback. Thank you.

arsharma1, 2018-12-11:
htf.measures is a decorator.  You should be able to remove your wrapper_my_measurements_decorator function and just move its contents up a level.

```
def define_voltage_measurements():
  signals_list_of_dicts = list(csv.DictReader(open(conf.signals_csv_filename)))
  new_measurements_list = []
  for sub_dict in signals_list_of_dicts:
            m_name = sub_dict['Signal_Name']  # One of the signal names is 'V_3P3'
            m_min = sub_dict['Min_Value']
            m_max = sub_dict['Max_Value']
            m_doc = sub_dict['Info']
            m = htf.Measurement(m_name).in_range(m_min, m_max).doc(m_doc).with_units('volt')
            new_measurements_list.append(m)
  return htf.measures(*new_measurements_list)

@define_voltage_measurements
def check_voltage_phase(test):
  ...
```

zamora18, 2018-12-11:
When I tried that, I got an `Invalid phase result` error.

arsharma1, 2019-01-02:
Looks like it's working.  Reopen this issue if you have any other questions.

closed 2019-01-02 by arsharma1