google/openhtf #844: webgui lack of documentation

mattvenn, 2019-01-03:
Hi,

I'd like an alternative to the webgui, as it's not suitable for our factory. Our factory is in China and we need a much simpler interface with just a start button and some basic feedback.

I'm not a front end person and have very little experience with html/javascript, so what I'd really like to do is replace the webgui with a python qt5 app. This would run on a raspberry pi with a 5" screen.

I've found where some URLs are defined by the station server:

https://github.com/google/openhtf/blob/54e7e77f37d65614d02c3171609e709a8ed21211/openhtf/output/servers/station_server.py#L571

But it's not clear how the webgui finds out the station id, what phases there are, how to subscribe etc.

I've also tried creating a new object that inherits from StationWatcher, and then registers a callback that can send test data over a socket. This seems to work OK, but I'm not able to get information about how many phases there are etc.

Any suggestions on how to achieve what I want? Is there a discussion group for users of openhtf? Is anyone using this in a non English speaking factory and have needed a simpler interface?

Thanks for any pointers!

mattvenn, 2019-01-03:
I got something working by using stationwatcher callbacks to emit signals into the pyqt5 framework
here for anyone interested: https://github.com/Dygmalab/test-jig/blob/master/openhtf/testjig.py

maxballenger, 2019-01-03:
Matt, nice work. I haven't dug into this too much myself but I imagine also
being interested in a very simple operator interface. Why do you prefer to
develop your interface in QT instead of web? I have always had difficulty
with QT as a finicky dependency and it seems to me that web interface tools
are always going to be ahead due to their orders of magnitude larger user
base and the standardization of browsers.

Max

mattvenn, 2019-01-04:
Hi Max,
yes it would probably be better to make a simpler front end for the webgui. However I don't have those skills and because we are a small team I need to be able to make changes fast myself. So I've gone with something I have more experience in. I use qtcreator to lay out a simple gui ATM, but because it is such a simple interface it could probably also be done programmatically, with each test having an additional decorator if it wants to be shown in the gui.

arsharma1, 2019-01-04:
You can start a new openhtf.servers.station_server.StationWatcher thread (https://github.com/google/openhtf/blob/master/openhtf/output/servers/station_server.py#L138) with a new callback function to hook into another GUI.  This callback function gets a basetypes version of TestState (https://github.com/google/openhtf/blob/master/openhtf/core/test_state.py#L104)

You can get a list of  all the phases by mimicking some of the code from PhaseHandler (https://github.com/google/openhtf/blob/master/openhtf/output/servers/station_server.py#L344):
```
def get_current_phases():
  test, _ = station_server._get_executing_test()  # pylint: disable=protected-access
  if test is None:
    return None
  return [
    dict(id=id(phase), **data.convert_to_base_types(phase))
    for phase in test.descriptor.phase_group
  ]
```

closed 2019-01-07 by mattvenn