google/openhtf #849: repeat a set of tests in a phase?

mattvenn, 2019-01-07:
Hi,

I have some helper methods to turn on and power to the DUT
Then I have test to read a firmware version, which is only available right after power is applied.

[power_off,
power_on,
read_firmware]

If the read_firmware fails, I'd like to use the phase REPEAT functionality.
If I put htf.PhaseResult.REPEAT inside read_firmware and limit retries to 3, then only that one test is repeated. Which doesn't work, because the power needs to be cycled.

Is there a way of using an inner phase group so that all 3 are repeated? I can always repeat the power control code inside the read_firmware test but I'd rather not.

arsharma1, 2019-01-07:
REPEAT is limited to a single phase only.  This would make the execution much more complex.

I'd recommend consolidating this into a single phase.  If the power control implementation is entirely handled in a Plug, it is easy to reference in the single phase.  It may help to create subclasses of an instrument-specific power plug specific to the device being controlled so that it knows exactly what parameters to use.

```
class InstrumentPowerPlug(plugs.BasePlug):
  def power_on(self):
    pass
  def power_off(self):
    pass

class MyDevicePowerPlug(InstrumentPowerPlug):
  # Includes overrides or configuration for powering the MyDevice with Instrument.
  pass

@plugs.plug(power=MyDevicePowerPlug, device=MyDevicePlug)
@htf.PhaseOptions(repeat_limit=3)
def read_firmware(test, power, device):
  power.power_off()
  power.power_on()
  try:
    firmware = device.read_firmware()
  except FailedFirmwareError:
    return htf.PhaseResult.REPEAT
  # Handle firmware...
```

mattvenn, 2019-01-08:
ah, cool! That's great. Then I can use the power plug in other related tests.

closed 2019-01-08 by mattvenn