google/openhtf #854: 'module' object has no attribute 'MIMETYPE_MAP' - station_server.py

zamora18, 2019-01-16:
I have openHTF installed in two locations but I only get this error on one of them (The newer one I installed today. I installed from source). I try running `frontend_example.py`, and this is the error I get:
```
Traceback (most recent call last):
  File "frontend_example.py", line 21, in <module>
    from openhtf.output.servers import station_server
  File "/home/lab_machine_2/anaconda2/lib/python2.7/site-packages/openhtf-1.3.0-py2.7.egg/openhtf/output/servers/station_server.py", line 36, in <module>
    v: k for k, v in six.iteritems(mfg_inspector.MIMETYPE_MAP)
AttributeError: 'module' object has no attribute 'MIMETYPE_MAP'
```
I am working on a project that also crashes on the new installation but works on a previously installed version.

scasagrande, 2019-01-21:
This looks to be related to the changes introduced in PR #848 

I am personally working off of the prior commit in master (4f7621c287e5902ba95ecd2cc919a90832e183b4) due to this.

Minipada, 2019-04-09:
Any news on this? @arsharma1

PR #870 (2019-07-26): fix: remove mention to mfg_inspector.MIMETYPE_MAP
Resolves #854

---
This change is [[image]](https://reviewable.io/reviews/google/openhtf/870)
commit 05b2a06a (2019-07-26): fix: remove mention to mfg_inspector.MIMETYPE_MAP (#870)

resolve #854
closed 2019-07-26 by arsharma1