google/openhtf #859: Frontend Example Broken

ashesman, 2019-02-27:
II cannot seem to get the frontend example to run.  Tried latest and V1.3 release.   The error output is quite different from both versions but both seem to be missing a dependency. 

It is highly possible this is user error due to my limited python knowledge so please disregard if it is! 

On latest:

ash@HD3:~/Delete/openhtf/examples$ python frontend_example.py 
Traceback (most recent call last):
  File "frontend_example.py", line 21, in <module>
    from openhtf.output.servers import station_server
  File "/home/ashleyduncan/Delete/openhtf/openhtf/output/servers/station_server.py", line 23, in <module>
    from openhtf.output.callbacks import mfg_inspector
  File "/home/ashleyduncan/Delete/openhtf/openhtf/output/callbacks/mfg_inspector.py", line 14, in <module>
    from openhtf.output.proto import test_runs_converter
  File "/home/ashleyduncan/Delete/openhtf/openhtf/output/proto/test_runs_converter.py", line 39, in <module>
    from openhtf.output.proto import test_runs_pb2
ImportError: cannot import name test_runs_pb2

On V1.3 Release

ash@HD3:~/Delete/openhtf/examples$ python frontend_example.py 
Traceback (most recent call last):
  File "frontend_example.py", line 21, in <module>
    from openhtf.output.servers import station_server
  File "/home/ashleyduncan/Delete/openhtf/openhtf/output/servers/station_server.py", line 26, in <module>
    from openhtf.output.callbacks import mfg_inspector
  File "/home/ashleyduncan/Delete/openhtf/openhtf/output/callbacks/mfg_inspector.py", line 48, in <module>
    from openhtf.output.proto import guzzle_pb2
ImportError: cannot import name guzzle_pb2

ashesman, 2019-02-27:
Looks like proto buffers were not built properly...

ashesman, 2019-02-28:
Sorry, still cannot get the frontend example to go at all.  Without using source code, just install package pip install open htp.  Copy front end example to empty folder and run it.  Cant load modules error.

RobertSte, 2019-05-29:
Same problem here any known fix?
One thing to note, I tried to install openhtf via pip before having the protobuf stuff installed and getting an error. After that I installed the two mentioned protobuf packages, deinstalled openhtf via pip and installed it in the virtualenv again without any errors. Maybe that has something to to with this problem.

RobertSte, 2019-06-06:
I could fix the problem with latest openHTF from pip (1.4.2), why get I this version by pip, it isn't a release in github, only a tag?

You could simply remove the lines  35..37 from .../lib/python3.x/site-packages/openhtf/output/servers/station_server.py. I guess that the MIMETYPE_REVERSE_MAP is just an undeleted artifact from an older (or future) version, because MIMETYPE_REVERSE_MAP has only one occurance in that file.

I'm trying to use openHTF for two or threes days now, and it's a shame that not even the included examples are tested for a release... Also I haven't found any documentation for a "user" so it's more a thing of guessing and trying to understand what is happening...
That's a bit of shame, because as far as I can see now, it is exactly what we need.