google/openhtf #863: How can I pass 'conf' variables to a plug?

git109, 2019-03-27:
I'm new to python and openhtf and I couldn't find any example on the repo on how to do it.  Please help.

```
import openhtf as htf
from openhtf.util import conf

conf.declare(
    'hw_io_line',
    description='gpio'
)

class HwTestPlug(htf.plugs.BasePlug):
    @conf.inject_positional_args
    def __init__(self, hw_io_line):
        self.hw_io_line = hw_io_line

@htf.plug(htp=HwTestPlug)
def hw_rf_test(test,htp):
    print("im here")

if __name__ == '__main__':
    test = htf.Test(hw_rf_test)
    test.execute(test_start=lambda: 'MyHw')
```

I've tried htf.Test(hw_rf_test).with_args(hw_io_line=23) but I couldn't get to work and I always get this error.

```	
Error: __init__() missing 1 required positional argument: 'hw_io_line'
```
It would be great if you can update the examples as this is a great framework.

arsharma1, 2019-03-27:
Test doesn't take with_args.

The conf.inject_position_args will automatically inject the parameters into your function.  Since your conf declaration does not have a default_value keyword argument, a value must be specified on the command line when running your program:

$ test.py --config-value=hw_io_line=123

However, I usually prefer to get the conf value directly, rather than using the conf.inject_positional_args decorator:
```
class HwTestPlug(htf.plugs.BasePlug):
  def __init__(self):
    self.hw_io_line = conf.hw_io_line
```

git109, 2019-03-27:
Thank you.  This is the info I was looking for.

kzhuo333, 2023-05-15:
Took me a while to figure this out and I would like to add more detail on what eventually worked for me.
Below is what I have in my serial plug file

`from openhtf.util import conf

conf.declare("serial", 'A dict that contains 3 keys, "comport" and "baudrate" and "timeout"')
conf.load_from_filename('config.yml')
conf.declare(name='port', default_value=conf.serial['comport'], description='comport')
conf.declare(name='baud', default_value=conf.serial['baudrate'], description='baud rate')
conf.declare(name='timeout', default_value=conf.serial['timeout'], description='timeout while waiting for comport response')

class serial_com(BasePlug):
    """Serial comport plug class."""

    @conf.inject_positional_args    
    def __init__(self, port, baud, timeout)->None:
        """Constructor for serial comms plug."""
        self.ser = serial.Serial()
        self.ser.baudrate = baud
        self.ser.port = port
        self.ser.timeout = timeout
`

My config file is like this

serial:
  comport: COM4
  baudrate : 9600
  timeout : 2