google/openhtf #864: independent plugs in same test

DigvijayPati, 2019-03-28:
Hi, I have two plugs (ea_psu_plug, motion_master_plug). below test gets stuck.
ideally, 1st phase should turn ON the power supply and rest phase runs actual tests with motor. 

- phase **test_start_psu** runs without any error and turns ON the power supply. 
_(note: phase test_position_control is commented out in this scenario)_

- But when i include phase **test_position_control** along with above phase then test gets stuck with log  _Executing phase trigger_phase_
  
```python
@htf.measures(htf.Measurement("psu_voltage").in_range(0, 48))
@htf.plug(epp=ea_psu_plug.EaPsuPlug)
def test_start_psu(test, output_num, voltage, epp):

    psu_found = epp.get_device_description(output_num)
    test.logger.info('power supply is %s', psu_found)
    #epp.remote_on(output_num)
    epp.set_voltage(voltage, output_num)

    epp.output_on(output_num)
    test.logger.info("power supply is turned ON")

    measured_voltage = epp.get_voltage(output_num)
    test.logger.info("psu voltage is set to %d V" % measured_voltage)
    test.measurements.psu_voltage = measured_voltage

@htf.measures(htf.Measurement("position_difference").in_range(900, 1100))
@htf.plug(mmp=motion_master_plug.MotionMasterPlug)
def test_position_control(test, node_id, mmp):

    actual_position = mmp.get_position(node_id)
    mmp.set_position(node_id, actual_position + 1000)
    time.sleep(1)
    test.measurements.position_difference = abs(mmp.get_position(node_id) -
                                                actual_position)
if __name__ == '__main__':

        test_name = "test_motor_control"
        dut = "single_node_chain"

        test = htf.Test(
            test_start_psu.with_args(output_num=0, voltage=48),
            test_position_control.with_args(node_id=0),
            test_name=test_name)

        test.execute(test_start=lambda: dut)
```

arsharma1, 2019-03-28:
What happens when you manually turn on power supply and run just the test_position_control phase?

Your description suggests to me that the MotionMasterPlug is expecting the device to be operational during its __init__ function.  I'd recommend moving that out to a new phase (or combine with test_start_psu).

DigvijayPati, 2019-03-28:
test_position_control phase runs successfully when I turns on power supply manually.

DigvijayPati, 2019-03-28:
yes, MotionMasterPlug's init function has separate function to scan ethercat device which is connected to power supply. This device needs to be turned ON first to get operational. But that requirement will be fulfilled by test_start_psu phase, right? Does openhtf initialize all the plugs in-parallel?

test_start_psu phase (with power supply off) & test_position_control phase  (with power supply ON) are running fine independently.  but I wonder, what is blocking when both phases are running one after?

cricdecyan, 2019-03-28:
Hi Digvijay,
How about you don't check from ethercat? You could be having a race
condition in your plug initialization because of that strange dependency
you have.

arsharma1, 2019-03-28:
Plug initialization takes place in two locations:
1) ** Initialize trigger phase plugs **
2) Run trigger phase
3) ** Initialize all remaining plugs **
3) Run other phases

Plugs are initialized sequentially in an arbitrary order.

This construction allows plugs to be shared between phases.

I'd reiterate my suggestion from earlier.  Pull the ethercat initialization from the plug constructor into a new phase and add that it in between the power on and position control phases:
```
@htf.plug(mmp=motion_master_plug.MotionMasterPlug)
def init_ethercat(mmp):
  mmp.init_ethercat()
```

DigvijayPati, 2019-03-29:
Hi @arsharma1, Thank you very much. It is working now. 

```python
@htf.measures(htf.Measurement("psu_voltage").in_range(46, 48))
@htf.plug(epp=ea_psu_plug.EaPsuPlug)
def test_start_psu(test, output_num, voltage, epp):
    psu_found = epp.get_device_description(output_num)
    test.logger.info('power supply is %s', psu_found)
    epp.set_voltage(voltage, output_num)
    epp.output_on(output_num)
    test.logger.info("power supply is turned ON")
    measured_voltage = epp.get_voltage(output_num)
    test.logger.info("psu voltage is set to %d V" % measured_voltage)
    test.measurements.psu_voltage = measured_voltage

@htf.measures(htf.Measurement("num_of_slaves")equals(1))
@htf.plug(mmp=motion_master_plug.MotionMasterPlug)
def search_ethercat_devices(test, mmp):
    num_of_slaves = mmp.scan_devices()
    mmp.add_devices(num_of_slaves)
    test.measurements.num_of_slaves = num_of_slaves
    time.sleep(2)

@htf.measures(htf.Measurement("position_difference").in_range(900, 1100))
@htf.plug(mmp=motion_master_plug.MotionMasterPlug)
def test_position_control(test, node_id, mmp):
    actual_position = mmp.get_position(node_id)
    mmp.set_position(node_id, actual_position + 1000)
    time.sleep(1)
    test.measurements.position_difference = abs(mmp.get_position(node_id) -
                                                actual_position)
if __name__ == '__main__':
        test_name = "test_motor_control"
        dut = "single_node_chain"
        test = htf.Test(
            test_start_psu.with_args(output_num=0, voltage=48),
            search_ethercat_devices
            test_position_control.with_args(node_id=0),
            test_name=test_name)
        test.execute(test_start=lambda: dut)
```
Can I include **PhaseGroup** if I don't want to continue subsequent phases if setup phase(turning on the power supply & ethercat initialization) fails?

DigvijayPati, 2019-04-15:
@arsharma1 : sometimes my test gets stuck after finishing its execution. Does openhtf tears down all plugs in parallel? Should I include delay in teardown function of power supply plug? How can I handle priority for shutting down plugs?

arsharma1, 2019-04-15:
@DigvijayPati 

An ERROR in a PhaseGroup will still stop the entire test run, which is the same for normal phases.

Teardown is sequential in an arbitrary order.  If you need to turn off devices in a particular order, that is where a PhaseGroup's teardown can help.  A PhaseGroup ensures that if its setup phases do not ERROR, then its teardown phases are guaranteed to run.

You can use it like this:

```
@htf.plug(mmp=motion_master_plug.MotionMasterPlug)
def shutdown_ethercat(test, mmp):
  mmp.shutdown()

...
  test = htf.Test(
    phase_group.PhaseGroup.with_teardown(shutdown_ethercat)(
            test_start_psu.with_args(output_num=0, voltage=48),
            search_ethercat_devices,
            test_position_control.with_args(node_id=0),
   ), test_name=test_name)
  ...
```

In this example , the teardown function will always be run since there are no setup functions.  If the shutdown requires the search to have happened, then rearrange as:

```
  test = htf.Test(
    phase_group.PhaseGroup(
        setup=[test_start_psu.with_args(output_num=0, voltage=48), search_ethercat_devices],
        main=[test_position_control.with_args(node_id=0)],
        teardown=[shutdown_ethercat],
   ), test_name=test_name)
```

DigvijayPati, 2019-04-18:
@arsharma1 : Thank you very much. I appreciate your support. I am able to implement phasegroups now. It is working perfectly now. any plans for next release?