google/openhtf #881: pip install error - windows 10

Hfusi0n, 2019-05-14:
I got the following error log while attempting to install from pip on windows 10:

Building wheels for collected packages: openhtf
  Building wheel for openhtf (setup.py) ... error
  ERROR: Complete output from command 'C:\ProgramData\Anaconda3\envs\DeviceQualityLibrary\python.exe' -u -c 'import setuptools, tokenize;__file__='"'"'C:\\Users\\org\\AppData\\Local\\Temp\\pip-install-gnbdz6nh\\openhtf\\setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' bdist_wheel -d 'C:\Users\org\AppData\Local\Temp\pip-wheel-cq903ymg' --python-tag cp37:
  ERROR: running bdist_wheel
  The [wheel] section is deprecated. Use [bdist_wheel] instead.
  running build
  running build_proto
  Traceback (most recent call last):
    File "<string>", line 1, in <module>
    File "C:\Users\org\AppData\Local\Temp\pip-install-gnbdz6nh\openhtf\setup.py", line 221, in <module>
      'pytest-cov>=2.2.1',
    File "C:\ProgramData\Anaconda3\envs\DeviceQualityLibrary\lib\site-packages\setuptools\__init__.py", line 145, in setup
      return distutils.core.setup(**attrs)
    File "C:\ProgramData\Anaconda3\envs\DeviceQualityLibrary\lib\distutils\core.py", line 148, in setup
      dist.run_commands()
    File "C:\ProgramData\Anaconda3\envs\DeviceQualityLibrary\lib\distutils\dist.py", line 966, in run_commands
      self.run_command(cmd)
    File "C:\ProgramData\Anaconda3\envs\DeviceQualityLibrary\lib\distutils\dist.py", line 985, in run_command
      cmd_obj.run()
    File "C:\ProgramData\Anaconda3\envs\DeviceQualityLibrary\lib\site-packages\wheel\bdist_wheel.py", line 192, in run
      self.run_command('build')
    File "C:\ProgramData\Anaconda3\envs\DeviceQualityLibrary\lib\distutils\cmd.py", line 313, in run_command
      self.distribution.run_command(command)
    File "C:\ProgramData\Anaconda3\envs\DeviceQualityLibrary\lib\distutils\dist.py", line 985, in run_command
      cmd_obj.run()
    File "C:\ProgramData\Anaconda3\envs\DeviceQualityLibrary\lib\distutils\command\build.py", line 135, in run
      self.run_command(cmd_name)
    File "C:\ProgramData\Anaconda3\envs\DeviceQualityLibrary\lib\distutils\cmd.py", line 313, in run_command
      self.distribution.run_command(command)
    File "C:\ProgramData\Anaconda3\envs\DeviceQualityLibrary\lib\distutils\dist.py", line 983, in run_command
      cmd_obj = self.get_command_obj(command)
    File "C:\ProgramData\Anaconda3\envs\DeviceQualityLibrary\lib\distutils\dist.py", line 858, in get_command_obj
      cmd_obj = self.command_obj[command] = klass(self)
    File "C:\ProgramData\Anaconda3\envs\DeviceQualityLibrary\lib\distutils\cmd.py", line 62, in __init__
      self.initialize_options()
    File "C:\Users\org\AppData\Local\Temp\pip-install-gnbdz6nh\openhtf\setup.py", line 70, in initialize_options
      maybe_protoc = os.path.join(prefix, 'bin', 'protoc')
  UnboundLocalError: local variable 'prefix' referenced before assignment
  Warning: mfg-inspector output is not fully implemented for Windows. OpenHTF will be installed without it.
  ----------------------------------------
  ERROR: Failed building wheel for openhtf

arsharma1, 2019-05-14:
This will need a new point release after the fix goes in.

Hfusi0n, 2019-06-05:
I was able to resolve this issue the following way: 
I've installed pkg-config lite from: 
https://sourceforge.net/projects/pkgconfiglite/

Next I set an environment variable PKG_CONFIG_PATH with a path to protobuf.pc file 
This file was located for me at: 
C:\Users\<username>\AppData\Local\conda\conda\pkgs\libprotobuf-3.7.1-h7bd577a_0\Library\lib\pkgconfig

A better solution should be integrated

ausafaq, 2019-11-13:
Hi, I am facing the same error. Please suggest a fix!

Hfusi0n, 2019-11-13:
> Hi, I am facing the same error. Please suggest a fix!

Hi,

Please find my above reply... you can install pkgconfiglite and set a new env variable:
PKG_CONFIG_PATH to protobuf.pc file

I think you should be able to get the file and the package with conda:
https://anaconda.org/anaconda/libprotobuf

A better solution should be comitted to the code resolving this issue...

arsharma1, 2019-11-13:
I'll accept a pull request with a proper fix.

ipfilip, 2022-09-01:
Hi, I also encountered this error. I'm supprised it's such an old issue.