google/openhtf #888: User input broken on raspbian, Ideas?

RobertSte, 2019-06-19:
I have openHTF up and running in a virtual machine running Ubuntu 18.04.2 LTS with Python 3.6.8 so far so good. The examples from the example folder are running without problems.

I like to use the software on a Raspberry Pi 3B running latest raspbian.
I tried python 3.5.3 (default on raspbian) and complied installed python 3.6.8 from source, both versions showing the same behavior/error.

Running hello_world_py:
```
(dhe) pi@openhtfpi:~/playground/openhtf/examples $ python --version
Python 3.6.8
(dhe) pi@openhtfpi:~/playground/openhtf/examples $ python hello_world.py 
enter prompt
Enter a DUT ID in order to start the test.
exit prompt
--> testpoint 2 - 86400
Traceback (most recent call last):
  File "hello_world.py", line 89, in <module>
    test.execute(test_start=user_input.prompt_for_test_start())
  File "/home/pi/python-environments/dhe/lib/python3.6/site-packages/openhtf/core/test_descriptor.py", line 349, in execute
    outcome=final_state.test_record.outcome.name,
AttributeError: 'NoneType' object has no attribute 'name'
```
The test doesn't wait at the user input and throws the error directly.
During debugging I added the "testpoint2 in user_input.py:
```
def wait_for_prompt(self, timeout_s=None):
    """Wait for the user to respond to the current prompt.

    Args:
      timeout_s: Seconds to wait before raising a PromptUnansweredError.

    Returns:
      A string response, or the empty string if text_input was False.

    Raises:
      PromptUnansweredError: Timed out waiting for the user to respond.
    """
    with self._cond:
      if self._prompt:
        if timeout_s is None:
          self._cond.wait(3600 * 24 * 365)
        else:
          print ('testpoint 2 - ' + str(timeout_s) )
          self._cond.wait(timeout_s)
          print ('testpoint 3 - ' + str(timeout_s) )
      if self._response is None:
        raise PromptUnansweredError
      print ('self._response' + str( self._response ) )
      return self._response
```
So my guess is, that the wait is working but somehow the user_input returns immediately on the Raspberry Pi, therefore the name is not set and the test runs into the exception.

Any ideas what is going wrong here. Must be soemething raspbian specific...

Best, Robert

RobertSte, 2019-06-19:
The raspi is running with full desktop, real keyboard and mouse and via HDMI on a display. No remote control no SSH, VNC or such stuff

arsharma1, 2019-06-19:
What version of openhtf did you install?  It's pulling the code from a
different location that the examples you're running.

RobertSte, 2019-06-20:
Hi,
we forked the openHTF git repo at cb544023e0d9acf758add54026d43668cc37e091
since I fixed the known bugs #854, #833, #887 by removing the three lines from station server.py and added some plugs and a csv factory output callback (which aren't used by the hello world example at all).
So the examples and the openhtf core are unchanged since the last commit here.
I also tried the openHTF version from this repo, same result as expected.

Best, Robert

arsharma1, 2019-06-21:
Can you post the stacktrace you get when you run with the latest OpenHTF version?

RobertSte, 2019-06-21:
Here it is:

```
pi@openhtfpi:~ $ ls
Desktop    Downloads  Music     playground  python-environments  Videos
Documents  MagPi      Pictures  Public      Templates
pi@openhtfpi:~ $ cd python-environments/
pi@openhtfpi:~/python-environments $ virtualenv -p python3 ori
Running virtualenv with interpreter /usr/local/bin/python3
Using base prefix '/usr/local'
New python executable in /home/pi/python-environments/ori/bin/python3
Also creating executable in /home/pi/python-environments/ori/bin/python
Installing setuptools, pkg_resources, pip, wheel...done.
pi@openhtfpi:~/python-environments $ source ori/bin/activate
(ori) pi@openhtfpi:~/python-environments $ pip install git+https://github.com/google/openhtf.gitLooking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Collecting git+https://github.com/google/openhtf.git
  Cloning https://github.com/google/openhtf.git to /tmp/pip-req-build-wjor9hmm
  Running command git clone -q https://github.com/google/openhtf.git /tmp/pip-req-build-wjor9hmm
Collecting colorama<1.0,>=0.3.9 (from openhtf==1.4.2)
  Using cached https://files.pythonhosted.org/packages/4f/a6/728666f39bfff1719fc94c481890b2106837da9318031f71a8424b662e12/colorama-0.4.1-py2.py3-none-any.whl
Collecting contextlib2<1.0,>=0.5.1 (from openhtf==1.4.2)
  Using cached https://files.pythonhosted.org/packages/a2/71/8273a7eeed0aff6a854237ab5453bc9aa67deb49df4832801c21f0ff3782/contextlib2-0.5.5-py2.py3-none-any.whl
Collecting future>=0.16.0 (from openhtf==1.4.2)
  Using cached https://www.piwheels.org/simple/future/future-0.17.1-py3-none-any.whl
Collecting mutablerecords<2.0,>=0.4.1 (from openhtf==1.4.2)
Collecting oauth2client<2.0,>=1.5.2 (from openhtf==1.4.2)
  Using cached https://www.piwheels.org/simple/oauth2client/oauth2client-1.5.2-py3-none-any.whl
Collecting protobuf<4.0,>=3.6.0 (from openhtf==1.4.2)
  Using cached https://files.pythonhosted.org/packages/8f/5a/077f2811869284d8e15fcc88818922e19cf205a3529f21210f3a1b47dc74/protobuf-3.8.0-py2.py3-none-any.whl
Collecting PyYAML<4.0,>=3.13 (from openhtf==1.4.2)
Collecting pyOpenSSL<18.0,>=17.1.0 (from openhtf==1.4.2)
  Using cached https://files.pythonhosted.org/packages/79/db/7c0cfe4aa8341a5fab4638952520d8db6ab85ff84505e12c00ea311c3516/pyOpenSSL-17.5.0-py2.py3-none-any.whl
Collecting sockjs-tornado<2.0,>=1.0.3 (from openhtf==1.4.2)
  Using cached https://www.piwheels.org/simple/sockjs-tornado/sockjs_tornado-1.0.6-py3-none-any.whl
Collecting tornado<5.0,>=4.3 (from openhtf==1.4.2)
Collecting pyasn1-modules>=0.0.5 (from oauth2client<2.0,>=1.5.2->openhtf==1.4.2)
  Using cached https://files.pythonhosted.org/packages/91/f0/b03e00ce9fddf4827c42df1c3ce10c74eadebfb706231e8d6d1c356a4062/pyasn1_modules-0.2.5-py2.py3-none-any.whl
Collecting pyasn1>=0.1.7 (from oauth2client<2.0,>=1.5.2->openhtf==1.4.2)
  Using cached https://files.pythonhosted.org/packages/7b/7c/c9386b82a25115cccf1903441bba3cbadcfae7b678a20167347fa8ded34c/pyasn1-0.4.5-py2.py3-none-any.whl
Collecting httplib2>=0.9.1 (from oauth2client<2.0,>=1.5.2->openhtf==1.4.2)
  Using cached https://files.pythonhosted.org/packages/a8/12/213dee680d0e7d5689e3ef20c857f72932964c9d369ec50fbc39906fc757/httplib2-0.13.0-py3-none-any.whl
Collecting rsa>=3.1.4 (from oauth2client<2.0,>=1.5.2->openhtf==1.4.2)
  Using cached https://files.pythonhosted.org/packages/02/e5/38518af393f7c214357079ce67a317307936896e961e35450b70fad2a9cf/rsa-4.0-py2.py3-none-any.whl
Collecting six>=1.6.1 (from oauth2client<2.0,>=1.5.2->openhtf==1.4.2)
  Using cached https://files.pythonhosted.org/packages/73/fb/00a976f728d0d1fecfe898238ce23f502a721c0ac0ecfedb80e0d88c64e9/six-1.12.0-py2.py3-none-any.whl
Requirement already satisfied: setuptools in ./ori/lib/python3.6/site-packages (from protobuf<4.0,>=3.6.0->openhtf==1.4.2) (41.0.1)
Collecting cryptography>=2.1.4 (from pyOpenSSL<18.0,>=17.1.0->openhtf==1.4.2)
Collecting cffi!=1.11.3,>=1.8 (from cryptography>=2.1.4->pyOpenSSL<18.0,>=17.1.0->openhtf==1.4.2)
Collecting asn1crypto>=0.21.0 (from cryptography>=2.1.4->pyOpenSSL<18.0,>=17.1.0->openhtf==1.4.2)
  Using cached https://files.pythonhosted.org/packages/ea/cd/35485615f45f30a510576f1a56d1e0a7ad7bd8ab5ed7cdc600ef7cd06222/asn1crypto-0.24.0-py2.py3-none-any.whl
Collecting pycparser (from cffi!=1.11.3,>=1.8->cryptography>=2.1.4->pyOpenSSL<18.0,>=17.1.0->openhtf==1.4.2)
  Using cached https://www.piwheels.org/simple/pycparser/pycparser-2.19-py2.py3-none-any.whl
Building wheels for collected packages: openhtf
  Building wheel for openhtf (setup.py) ... done
  Stored in directory: /tmp/pip-ephem-wheel-cache-ijrtxd4h/wheels/9a/23/59/5fb599f346e3aa060ddd797f6be8bcbb681f32d9eecba8f552
Successfully built openhtf
Installing collected packages: colorama, contextlib2, future, mutablerecords, pyasn1, pyasn1-modules, httplib2, rsa, six, oauth2client, protobuf, PyYAML, pycparser, cffi, asn1crypto, cryptography, pyOpenSSL, tornado, sockjs-tornado, openhtf
Successfully installed PyYAML-3.13 asn1crypto-0.24.0 cffi-1.12.3 colorama-0.4.1 contextlib2-0.5.5 cryptography-2.7 future-0.17.1 httplib2-0.13.0 mutablerecords-0.4.1 oauth2client-1.5.2 openhtf-1.4.2 protobuf-3.8.0 pyOpenSSL-17.5.0 pyasn1-0.4.5 pyasn1-modules-0.2.5 pycparser-2.19 rsa-4.0 six-1.12.0 sockjs-tornado-1.0.6 tornado-4.5.3
(ori) pi@openhtfpi:~/python-environments $ python --version
Python 3.6.8
(ori) pi@openhtfpi:~/python-environments $ cd ..
(ori) pi@openhtfpi:~ $ git clone https://github.com/google/openhtf.git
Cloning into 'openhtf'...
remote: Enumerating objects: 16, done.
[... cut ...]
  File "hello_world.py", line 89, in <module>
    test.execute(test_start=user_input.prompt_for_test_start())
  File "/home/pi/python-environments/ori/lib/python3.6/site-packages/openhtf/core/test_descriptor.py", line 348, in execute
    outcome=final_state.test_record.outcome.name,
AttributeError: 'NoneType' object has no attribute 'name'
(ori) pi@openhtfpi:~/openhtf/examples $ strace python hello_world.py &> stacktrace.txt
```
[stacktrace.txt](https://github.com/google/openhtf/files/3313422/stacktrace.txt)

arsharma1, 2019-06-21:
How are you connected to the Pi?  SSH or Serial?

What does `sys.stdin.isatty()` return?

RobertSte, 2019-06-24:
As written in my second post, I have attached a real keyboard an mouse (via USB) to Raspberry Pi.
The Pi is running the full Desktop and is connected via HDMI to the Display. So the Pi is running as a normal "Desktop" PC.
No SSH, no VNC, no Serial connections.

justinsg, 2019-07-07:
I have the exact same issue.

I've narrowed the issue down to the [test executor wait() function](https://github.com/google/openhtf/blob/master/openhtf/core/test_executor.py#L115-L123), which returns almost immediately. It seems that the thread join times out (i.e. `thread.join()` returns but an immediate `thread.is_alive()` check shows the thread is still running).

I think it's related to the thread join timeout, because the issue goes away if I revert a6083642d672 - i.e. set the thread join timeout to `31557600`, rather than `threading.TIMEOUT_MAX`.

The value of `TIMEOUT_MAX` on the Pi is `9223372036.0`. With a bit of trial and error, I can set a timeout of up to about `584950000.0` before the issue starts occurring intermittently. I'm not seeing any `OverflowError` thrown either which is odd.

@RobertSte can you check your setup to see if it responds similarly to different timeout values?

FYI:

On my Raspberry Pi:
```
Python 3.5.3 (default, Sep 27 2018, 17:25:39) 
[GCC 6.3.0 20170516] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import threading
>>> threading.TIMEOUT_MAX
9223372036.0
```

On my Ubuntu 18.04 workstation:
```
Python 3.6.8 (default, Jan 14 2019, 11:02:34) 
[GCC 8.0.1 20180414 (experimental) [trunk revision 259383]] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import threading
>>> threading.TIMEOUT_MAX
9223372036.0
```

arsharma1, 2019-07-09:
This sounds plausibly like a 32-bit overflow issue.  The Python runtime adds the the timeout to the current time in a time_t variable, which is likely a signed 32-bit number.  The `9223372036.0` timeout plus the current unix time winds up negative, so the internal wait finishes immediately.  `584950000.0` is also just enough more than now to also overflow to negative.

I'll revert the offending commit.

Thanks @justinsg for the additional details.

arsharma1, 2019-07-09:
The original fix was for Windows compatibility, so working around that with a new change.  Please pull #891 and let me know if that helps.

justinsg, 2019-07-09:
Yup #891 solves the problem for me, and your explanation makes sense too. Thanks @arsharma1 !

arsharma1, 2019-07-10:
I merged the timeout fix.  @RobertSte please let me know if this fixes your issue.

moradel82, 2019-10-09:
Hello I'm starting to use openhtf in a raspberry pi and I'm seen the same errors. I had to the small timeout in the test_descriptor to avoid the issue. Is there any other fixes for this?

arsharma1, 2019-10-09:
Hi @moradel82, are you using the PyPI release or have you pulled the latest Github changes?  I expect it to be fixed with the latest Github changes.

moradel82, 2019-10-10:
I used pip to install it.

arsharma1, 2019-10-10:
I'll push a new version with the fix up shortly, then.

arsharma1, 2019-10-14:
I've pushed 1.4.4 to PyPI.  Please let me know if you continue to have issues.

moradel82, 2019-10-17:
> I've pushed 1.4.4 to PyPI. Please let me know if you continue to have issues.

Thank you it does work now.

closed 2020-02-24 by arsharma1