google/openhtf #901: Question: Where to throw exception to FAIL or ERROR test

tcinbis, 2019-11-12:
This question is somehow related to #665 and #365.

Let's assume we use a plug which performes and returns some sort of measurement. This plug is then used in a monitor which in turn is added to a test phase by using the `monitors` decorator.

In case the plug fails to measure or encounters another exception, where and how should an exception be raised and handled such that the tests which use the monitor also fails?

If I just raise an exception in the plug (similar for inside a monitor) it will be visible in the log, but the test still will not be notified about it. It may encounter errors when trying to process values from the monitor though. The worst-case would be that the plug fails more or less silently and a test still passes.

Example code:
```python
class SomePlug(plugs.BasePlug):
    def __init__(self):
        super(SomePlug, self).__init__()

    def perform_measurement(self):
        # raise exception here?
        raise Exception # or just signal a failure by returning special value
```
```python
@htf.plug(some_plug=SomePlug)
def monitor_some_plug(some_plug):
    measurement = some_plug.perform_measurement()
    if measurement == INVALID_MEASUREMENT:
        # raise exception here?
        raise Exception
    return measurement
```

```python
@htf.monitors(
    "monitor_some_plug",
    monitor_some_plug,
    poll_interval_ms=500,
)
def some_test(test):
    time.sleep(5)
    # check results of monitor here 
    # raise exception or handle it?
```

arsharma1, 2019-11-12:
It looks like the monitor wrapping code does not catch or save exceptions.  An exception during the monitoring will cause the background thread to stop, so no further measurements would be taken.  It doesn't report an error or cause the test to fail.

One way to handle it for now would be to add a special bad value to your monitor function that either:
1) a measurement validator checks for and causes a failure
or
2) a followup phase checks for to cause an ERROR to stop the test.

tcinbis, 2019-11-19:
@arsharma1 thanks for your reply! I wanted to make sure that I do not miss an existing mechanism which already takes care of such things. So now I will use your first suggestion.

closed 2019-11-19 by tcinbis