google/openhtf #904: Setting validator targets during test execution

JamesMTSloan, 2019-12-01:
My use-case is setting a power supply voltage to a target value, and then checking that the voltage has been successfully achieved within a given tolerance. The target value is only known at runtime and is not hardcoded in the hardware tests (it could be but I want to explore this option). I have come up with a couple of ways of implementing this and wanted to ask for some thoughts here.

An example use-case:
```python
# at this point the limits are unknown
@measures(Measurement('powersupply_a_min', validators=[InRange(???)]))
def test_powersupply_a_minimum(test):
    # at this point the limits are available but there is no easy
    # mechanism to specify them as a validator
    # global my_validator
    target_voltage = get_target_voltage()
    set_supply_a(target_voltage)
    test.measurements.powersupply_a_min = measure_supply_a()
```

The following are methods of implementing this:

1. Accessing the internal `_measurements` dictionary of `Collection` and adding a new validator at during the test: `test.measurements._measurements['voltage'].with_validator(InRange(minimum=4.485, maximum=4.515))`

2. Adding a method to `Collection` that can add validators during the test.

3. Using a validator class with a hacked `__deepcopy__()` method and setting it as normal using the decorator and manipulating it as a module-scope variable.

Method 1 is obviously using the internals of `Collection` and so is not desirable but it demonstrates possible access. This gives rise to the idea of having measurements returning their value via `test.measurements.measurement_name` when the value has been set, and prior to that returning the `Measurement` object so that it can be manipulated (in this case its `with_validator()` method would be called.

Method 2 is simple and works (as here https://github.com/JamesMTSloan/openhtf/commit/b59a5fe33bcd5324cf4df0afaf92a2de9b79fb19) but I suppose it rather violates what you want `Collection` to do by introducing public methods to it.

Method 3 is a nasty hack demonstrated as follows to around the deep copying that OpenHTF does when setting up the execution threads:
```python
class RunTimeTarget(ValidatorBase):
    def __init__(self):
        self.target = None

    def __call__(self, value):
        if self.target is None:
            raise Exception("Target has not yet been set for this validator.")
        return self.target == value

    def __deepcopy__(self, memodict={}):
        return self

my_validator = RunTimeTarget()

@measures(Measurement('something', validators=[my_validator]))
def measure_something(test):
  my_validator.target = 5
  test.measurements.something = 5
```

I do like the syntax though of using the walrus operator to make your validator accessible within the test.
```python
@measures(Measurement('something', validators=[my_validator := RunTimeTarget()]))
```

For my own use-case I will probably make sure the targets are available when defining the tests but perhaps there are cases where this is not possible. Is this just a use-case that should always be discouraged as it is too dynamic for hardware testing?

hatharry, 2019-12-30:
I'm currently testing Method 1. The only problem I found was that the web-gui stops updating values in real-time, the phase completes then the values show.

JamesMTSloan, 2020-01-15:
Another suggestion for how to implement this is the following.

A simple class used to collect the measured values and any further validators to be applied (in addition to any already added normally);
```python
class MeasurementValueWithValidators:
  def __init__(self, measurement_value, *validators):
    self.measurement_value = measurement_value
    self.validators_to_set = validators
```

Then by modifying `Collection.__setitem__()` as follows https://github.com/google/openhtf/blob/master/openhtf/core/measurements.py#L595:
```python
  def __setitem__(self, name, value):  # pylint: disable=invalid-name
    self._assert_valid_key(name)
    if self._measurements[name].dimensions:
      raise InvalidDimensionsError(
          'Cannot set dimensioned measurement without indices')
    if isinstance(value, MeasurementValueWithValidators):
      for v in value.validators_to_set:
        self._measurements[name].with_validator(v)
      value = value.measurement_value
    self._measurements[name].measured_value.set(value)
    self._measurements[name].notify_value_set()
```

You can then record a measurement whilst adding new validators like so:
```python
@measures(Measurement('powersupply_a_min'))
def test_powersupply_a_minimum(test):
    target_voltage = get_target_voltage()
    set_supply_a(target_voltage)
    result = measure_supply_a()
    new_validator = InRange(minimum=4.485, maximum=4.515)
    test.measurements.powersupply_a_min = MeasurementValueWithValidators(result, new_validator)
```
But the normal syntax for recording measurements is still maintained.

tcm0116, 2020-02-29:
It seems like the issue I've described in #925 regarding with_plugs.py is related to this issue in that I believe that example would work for your use case if it could be determined why it's not actually working.