google/openhtf #935: StationServer doesn't record history

karlp, 2020-06-12:
Am I correct in reading that the StationServer's history_path argument is just to point to where "something else" is actually writing files for history objects?

It appears that the StationServer is only using PubSub to publish results and the [HistoryListHandler](https://github.com/google/openhtf/blob/master/openhtf/output/servers/station_server.py#L409) seems to actually be configured to expect some out of tree system that is writing serialized protobufs?

The reference "chtf.py" is ... "elsehere" ?  Does one have to build all this yourself to be able to use the features of StationServer or am I missing the pieces to connect?

Zeebrommer, 2020-06-26:
I am also interested in getting the station history to work. The readme mentions [a rundir flag](https://github.com/google/openhtf/blob/master/README.md#station) but i can't find where to apply that. The designated history folder remains empty when running tests.

karlp, 2020-06-26:
See https://github.com/google/openhtf/pull/936 for an implementation