google/openhtf #953: start_time_millis keeps initial value when running test executor in a loop

wallacbe, 2020-09-24:
start_time_millis keeps initial value when running test executor in a loop

wallacbe, 2020-09-27:
Here's script that can reproduce:

```
import openhtf
from openhtf.plugs import user_input
from openhtf.output.callbacks import mfg_inspector
from openhtf.output.proto import test_runs_converter

FILENAME_PATTERN = '{dut_id}-{metadata[test_name]}-{start_time_millis}.pb'

def phase(test):
  pass

test = openhtf.Test(phase)

test.add_output_callbacks(mfg_inspector.MfgInspector().set_converter(
    test_runs_converter.test_run_from_test_record).save_to_disk(
        FILENAME_PATTERN))

while True:
  test.execute(test_start=user_input.prompt_for_test_start())
~                                                               
```

wallacbe, 2020-09-28:
Is appears to be caused by the "cached proto" being returned.  What's the purpose of the cached proto?

https://github.com/google/openhtf/blob/9287d9d0df1588d4c794b33d33ef9a326b8127e7/openhtf/output/callbacks/mfg_inspector.py the te

```
  def _convert(self, test_record_obj):
    """Convert and cache a test record to a mfg-inspector proto."""

    if self._cached_proto is None:
      self._cached_proto = self._converter(test_record_obj)

    return self._cached_proto
```

arsharma1, 2020-09-28:
The cache looks the prevent converting multiple times when the code calls the save_to_disk and upload callbacks.

The cache could be made a little better if it kept track of the dut_id, station_id, and start_time_millis for the cached test record.  If those change, it could then regenerate the cache.

wallacbe, 2020-09-28:
Thanks for feedback, I can make that update

wallacbe, 2020-10-08:
PR is out for review https://github.com/google/openhtf/pull/955

closed 2020-10-08 by wallacbe