google/openhtf #963: TypeError: re.compile(<regular_expression>) has type re.Pattern, but expected one of: bytes, unicode

kunaljdoshi, 2020-10-27:
TypeError received when test_run_converter tries to create a protobuf file from a test_record which has a   compiled regular expression used in a validator with type: matches_regex

Traceback (most recent call last):
  File "/home/kunaldoshi/source/openhtf/openhtf/core/test_descriptor.py", line 348, in execute
    output_cb(final_state.test_record)
  File "/home/kunaldoshi/source/openhtf/openhtf/output/callbacks/mfg_inspector.py", line 215, in save_to_disk_callback
    proto = self._convert(test_record_obj)
  File "/home/kunaldoshi/source/openhtf/openhtf/output/callbacks/mfg_inspector.py", line 197, in _convert
    self._cached_proto = self._converter(test_record_obj)
  File "/home/kunaldoshi/source/openhtf/openhtf/output/proto/test_runs_converter.py", line 335, in test_run_from_test_record
    mangled_parameters = _extract_parameters(record, testrun,
  File "/home/kunaldoshi/source/openhtf/openhtf/output/proto/test_runs_converter.py", line 245, in _extract_parameters
    testrun_param.expected_text = validator.regex
TypeError: re.compile('[0-9A-Z]{4}') has type re.Pattern, but expected one of: bytes, unicode

kunaljdoshi, 2020-12-14:
Agreed. But that severely restricts one to use simple string comparisons when using matches_regex validators.

The change I am proposing is fairly limited in scope but lets a developer pass pre-compiled regex for comparision:
https://github.com/google/openhtf/pull/964/files

glados-verma, 2022-10-12:
To keep things simple (not just in the implementation, but also for measurement definitions, as well as handling in the UI and in outputs), we only support string regexes.

closed 2022-10-13 by glados-verma