google/openhtf #971: Cant bind multicast socket on Windows

ashesman, 2020-11-24:
When using frontend_example.py on Windows 10, I am not sure why but an exception is thrown on line 108 of multicast.py when trying to bind to 239.1.1.1 on port 10000.

`self._sock.bind((self.address, self.port))`

The exception information is below:

(env) D:\openhtf> cd d:\openhtf && cmd /C "d:\openhtf\env\Scripts\python.exe c:\Users\Ashley\.vscode\extensions\ms-python.python-2020.11.371526539\pythonFiles\lib\python\debugpy\launcher 60785 -- d:\openhtf\examples\frontend_example.py "
Exception in thread Thread-7:
Traceback (most recent call last):
  File "c:\users\ashley\appdata\local\programs\python\python37\lib\threading.py", line 926, in _bootstrap_inner
    self.run()
  File "d:\openhtf\openhtf\util\multicast.py", line 108, in run
    self._sock.bind((self.address, self.port))
OSError: [WinError 10049] The requested address is not valid in its context

ashesman, 2020-11-24:
Looks like this issue has a possible solution:[https://github.com/JamesMTSloan/openhtf/commit/0f3ab6ac916bf51613c0df5a3030a8dffbf5a7dc](https://github.com/JamesMTSloan/openhtf/commit/0f3ab6ac916bf51613c0df5a3030a8dffbf5a7dc)

dieser-niko, 2024-06-17:
It looks like Google doesn't seem to care about this sort of thing. However, I still want to use the official library. 
For that reason I'd recommend just changing the configuration. 
My code is based on the [frontend_example.py](https://github.com/google/openhtf/blob/master/examples/frontend_example.py) file:

```python
...

def main():
    CONF.load(station_server_port='4444',
              station_discovery_address="")  # added this line
    with station_server.StationServer() as server:
        web_launcher.launch('http://localhost:4444')
...
```