google/openhtf #979: Need Advice on how to Architect a Test Plan

sbluatx, 2021-01-30:
Folks, i have a question on how to architect a test plan for the following scenario:

[trigger] -> [set of tests] ->Pass/Fail->[if Pass Do A, else Do B. if ABORT Stop, else loop back]
                           ^------------------------------------------------------------------------------|
I have a a set of phases i need to run that would collectively in the box called [set of tests], then based on the results, i have another test node that does something specific, indicated by [if Pass Do A, if Fail Do B...]; then I go back to running [set of tests] unless I determine that i need to abort, in which case i abort out.

Can you please advice me on how I can setup such a scenario using openhtf?

cloudguruab, 2021-05-17:
I would like to contribute what would be a great place to start @sbluatx

sbluatx, 2021-05-18:
This was directed to the folks that developed the tool.  i'm wondering if this tool already supports the indicated scenario

cloudguruab, 2021-05-18:
@sbluatx no worries! I wouldn't mind giving it a try from what I can find on the docs & a few other sources.

glados-verma, 2021-07-08:
I haven't used them myself, but PhaseBranches are supposed to do what you've described.

closed 2022-10-12 by glados-verma